kizniche/Mycodo #1288: Unable to upgrade from 8.14.2

tomdotorg, 2023-03-24:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*after being off for 3 months, attempting to upgrade to the latest version prior to an settings export and then import to a new image*

### Versions:

 - Mycodo Version: * 8.14.2 *
 - Raspberry Pi Version: *Raspberry Pi 4 Model B Rev 1.4 *
 - Raspbian OS Version: *[Raspbian GNU/Linux 11 (bullseye)]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *Go to Upgrade*

### Expected behavior

*Able to upgrade*

### Screenshots

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 4 Model B Rev 1.4

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/tom/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 279, in decorated_view
    return func(*args, **kwargs)
  File "/home/tom/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

### Additional context

**

kizniche, 2023-03-24:
Duplicate of #1266

Please use the search feature before opening an issue. The fist sentence of the bug report template you're using explicitly states this.

closed 2023-03-24 by kizniche

tomdotorg, 2023-03-24:
Sorry about that.