kizniche/Mycodo #1291: Py code is not saving if "Analyze Python with Pylint" is unchecked

wingsergey, 2023-03-28:
### Describe the problem/bug

If uncheck 'Analyze Python with Pylint' flag during saving of Conditional Functions code under Run Python Code section is not applied in controller.

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *4B*
 - Raspbian OS Version: *Buster Lite*

### Additional context

In daemon logs I also can see old code in debug message

kizniche, 2023-03-28:
What version? Have you upgraded to master to confirm the unreleased version hasn't fixed this?

kizniche, 2023-03-28:
I see the issue at https://github.com/kizniche/Mycodo/blob/e76dbd7b4d65174e2ec12615943bcd2efaceca02/mycodo/mycodo_flask/utils/utils_conditional.py#L40

Thanks for the bug report. I'll commit a fix soon.

kizniche, 2023-03-28:
Upon further inspection, it is saving the code, it's just not providing an output.

closed 2023-04-06 by kizniche