kizniche/Mycodo #1292: Setup problem Mycodo not working 8.15.7

byhlk, 2023-04-01:
Versions:

 - Mycodo Version: *[8.15.7]*
 - Raspberry Pi Version: *[ZERO W]*
 - Raspbian OS Version: *[Pi OS Lite 32bit, bullseye, kernel 5.15.84]*

Trying to install mycodo on Raspberry Pi OS Lite, raspberry pi zero w hardware. I have domoticz, pi-hole and mikrotik hardware I thought it was a network related problem. I did android hot spot over LTE and still the same error. Below is the result of the installation.
Changed memory cards and different versions of os raspberry nothing helps. 

4. *[See error...]*
` https://project-downloads.drogon.net/wiringpi-latest.deb
Resolving project-downloads.drogon.net (project-downloads.drogon.net)... 188.246.205.22, 2a03:9800:10:7b::2
Connecting to project-downloads.drogon.net (project-downloads.drogon.net)|188.246.205.22|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-04-01 17:38:00 ERROR 404: Not Found.

dpkg-deb: error: unexpected end of file in archive magic version number in wiringpi-latest.deb
dpkg: error processing archive wiringpi-latest.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 wiringpi-latest.deb

#### Ensuring compatible version of influxdb 1.x is installed ####
#### Incorrect InfluxDB version (v(none)) installed. Should be v1.8.10-1
#### Stopping influxdb 2.x (if installed)...
Failed to stop influxd.service: Unit influxd.service not loaded.
#### Uninstalling influxdb 2.x (if installed)...`

`Apr 01 18:01:07 raspberrypi systemd[1]: Starting Mycodo server...
pi@raspberrypi:~ $ sudo service mycodo status
● mycodo.service - Mycodo server
     Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
     Active: activating (start-pre) since Sat 2023-04-01 18:01:07 BST; 1h 29min ago
Cntrl PID: 8817 (bash)
      Tasks: 2 (limit: 415)
        CPU: 2min 23.989s
     CGroup: /system.slice/mycodo.service
             ├─ 700 sleep 1
             └─8817 /bin/bash /var/mycodo-root/install/influxdb_wait_until_running.sh

Apr 01 18:01:07 raspberrypi systemd[1]: Starting Mycodo server...`

In the attachments log
[mycodokeepup.log](https://github.com/kizniche/Mycodo/files/11130279/mycodokeepup.log)
[mycodorestore.log](https://github.com/kizniche/Mycodo/files/11130280/mycodorestore.log)
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/11130282/mycodoupgrade.log)
[setup.log](https://github.com/kizniche/Mycodo/files/11130283/setup.log)
[login.log](https://github.com/kizniche/Mycodo/files/11130284/login.log)
[mycodo.log](https://github.com/kizniche/Mycodo/files/11130285/mycodo.log)
[mycodobackup.log](https://github.com/kizniche/Mycodo/files/11130286/mycodobackup.log)
[mycododependency.log](https://github.com/kizniche/Mycodo/files/11130287/mycododependency.log)
[mycodoimport.log](https://github.com/kizniche/Mycodo/files/11130288/mycodoimport.log)

kizniche, 2023-04-01:
Thank you for the bug report and attaching the setup log. Looks like the issue was influxdb not starting:

```
#### Flux found to not be enabled. Enabling and restarting InfluxDB. Job for influxdb.service failed because a timeout was exceeded.
```

See if influxdb will start again with:

`sudo service influxdb start`

byhlk, 2023-04-01:
**sudo service influxdb start** 

```
influxdb.service - InfluxDB is an open-source, distributed, time series database
     Loaded: loaded (/lib/systemd/system/influxdb.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: timeout) since Sat 2023-04-01 19:56:10 BST; 54ms ago
       Docs: https://docs.influxdata.com/influxdb/
    Process: 6870 ExecStart=/usr/lib/influxdb/scripts/influxd-systemd-start.sh (code=killed, signal=TERM)
        CPU: 1min 16.434s
```

**sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py**

```
2023-04-01 19:53:49,021 Mycodo daemon v8.15.7 starting
2023-04-01 19:53:49,334 Starting Pyro5 daemon
Object <__main__.PyroServer object at 0xb4612718>:
    uri = PYRO:mycodo.pyro_server@0.0.0.0:9080
Pyro daemon running.
2023-04-01 19:53:57,930 Activated in 4511.5 ms
2023-04-01 19:53:58,432 All activated Conditional controllers started
2023-04-01 19:53:58,435 All activated Trigger controllers started
2023-04-01 19:53:58,445 All activated Input controllers started
2023-04-01 19:53:58,460 All activated PID controllers started
2023-04-01 19:53:58,464 All activated Function controllers started
2023-04-01 19:54:00,078 Activated in 1608.4 ms
2023-04-01 19:54:01,082 Mycodo daemon started in 12.057 seconds
2023-04-01 19:54:03,607 47.42 MB RAM in use
```

kizniche, 2023-04-01:
The issue is influxdb wasn't properly set up, so you need to run install/setup.sh again. You may need to add a pause after that influxdb restart, before the influxdb dB and user creation is executed.

byhlk, 2023-04-01:
When I restart the system the influxDB does not turn on Daemon also. 
[image]

byhlk, 2023-04-01:
> p

How do you make it pause?

kizniche, 2023-04-01:
sleep 30

byhlk, 2023-04-01:
I don't know if I got it right but I changed the sleep value in the file 

**influxdb_wait_until_running.sh**
```
#!/bin/bash
# Sleeps until able to connect to InfluxDB (port 8086)
# Used by mycodo.service to determine when it's safe to start the Mycodo daemon

until nc -z localhost 8086; do sleep 30; done
```

Should I add sleep in the setup.sh installation file?
If so please specify where. Thank you in advance for your help

kizniche, 2023-04-01:
No. You only need to add the sleep between influxdb install and setup, after this:

https://github.com/kizniche/Mycodo/blob/1f480c0572e7b4538d7214059803db811d94e231/install/setup.sh#L193

byhlk, 2023-04-01:
I have no idea if I did it right, I don't know anything about it but I did it like this.

```
 ${INSTALL_CMD} update-influxdb-1 2>&1 | tee -a "${LOG_LOCATION}"
    sleep 60
    ${INSTALL_CMD} update-influxdb-1-db-user 2>&1 | tee -a "${LOG_LOCATION}"
```
Unfortunately it still doesn't connect to the database during installation

```
#### Creating InfluxDB 1.x database and user
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 60 seconds then trying again...
```

byhlk, 2023-04-01:
Before reinstalling, I uninstalled influxdb. 

~/Mycodo/install/setup.sh

Now there is this error

```

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package influxdb2
E: Unable to locate package influxdb2-cli
#### Installing InfluxDB v1.8.10-1...
Selecting previously unselected package influxdb.
(Reading database ... 46300 files and directories currently installed.)
Preparing to unpack influxdb_1.8.10_armhf.deb ...
Unpacking influxdb (1.8.10-1) ...
Setting up influxdb (1.8.10-1) ...
Processing triggers for man-db (2.9.4-2) ...
Job for influxdb.service failed because a timeout was exceeded.
See "systemctl status influxdb.service" and "journalctl -xe" for details.
Flux is already enabled.

#### Creating InfluxDB 1.x database and user
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 60 seconds then trying again...
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 60 seconds then trying again...
```

byhlk, 2023-04-02:
It works! I changed two entries.

**~/etc/systemd/system/multi-user.target.wants/influxdb.service**  add TimeoutStartSec=240

```
[Unit]
Description=InfluxDB is an open-source, distributed, time series database
Documentation=https://docs.influxdata.com/influxdb/
After=network-online.target

[Service]
User=influxdb
Group=influxdb
LimitNOFILE=65536
EnvironmentFile=-/etc/default/influxdb
ExecStart=/usr/lib/influxdb/scripts/influxd-systemd-start.sh
KillMode=control-group
Restart=on-failure
Type=forking
PIDFile=/var/lib/influxdb/influxd.pid
TimeoutStartSec=240
```

And what you suggested to me
https://github.com/kizniche/Mycodo/issues/1292#issuecomment-1493140729

After reinstalling from the location
~/Mycodo/install/setup.sh

Everything works! Thanks for your help. 
Topic to be closed...

byhlk, 2023-04-03:
I was too happy too soon. After a reboot, the raspberry unfortunately does not work.

influx
```
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: connect: connection refused
Please check your connection settings and ensure 'influxd' is running.
```
sudo service mycodo status

```
 sudo service mycodo status
● mycodo.service - Mycodo server
     Loaded: loaded (/home/byhlk/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
     Active: activating (start-pre) since Mon 2023-04-03 13:11:52 BST; 8min ago
Cntrl PID: 1716 (bash)
      Tasks: 2 (limit: 415)
        CPU: 14.016s
     CGroup: /system.slice/mycodo.service
             ├─1716 /bin/bash /var/mycodo-root/install/influxdb_wait_until_running.sh
             └─3940 sleep 1

Apr 03 13:11:52 raspberrypi systemd[1]: Starting Mycodo server...
```

When I give the command manually then influxdb not turns on

**sudo service influxdb start**

```
Job for influxdb.service failed because a timeout was exceeded.
See "systemctl status influxdb.service" and "journalctl -xe" for details.
```

byhlk, 2023-04-03:
I increased the time and it works :D

~/etc/systemd/system/multi-user.target.wants/influxdb.service add TimeoutStartSec=600

I increased the time and it works :D It starts up every time.

kizniche, 2023-04-06:
Seems to be the same issue as #1275