kizniche/Mycodo #1294: WiringPi url is no longer active

KShips, 2023-04-03:
Error during installs that it cannot find deb file, original project is no longer maintained. 

Invalid url in upgrade_commands.sh file:
WIRINGPI_URL="https://project-downloads.drogon.net/wiringpi-latest.deb"

Found this fork which looks to be drop-in replacement? (Have not tested):
https://github.com/WiringPi/WiringPi
WIRINGPI_URL="https://github.com/WiringPi/WiringPi/releases/download/2.61-1/wiringpi-2.61-1-armhf.deb"

kizniche, 2023-04-03:
Just pushed a potential fix, if you would like to test.

KShips, 2023-04-03:
I cannot see any reference to wiringpi when doing update on Master (should this be the case or only on fresh install?), running gpio -v in terminal comes up with command not found which I think indicates not installed.
 
[UpdateLog.txt](https://github.com/kizniche/Mycodo/files/11135875/UpdateLog.txt)

kizniche, 2023-04-06:
I don't believe wiringpi is reinstalled when upgrading, only when installing. I just released 8.15.8 with has this fix. I'm going to close the issue. If you find the fix doesn't work, feel free to reopen the issue.

closed 2023-04-06 by kizniche