kizniche/Mycodo #1299: Change I2C address TRI-PUMP Atlas Scientific

pe2269, 2023-04-17:
Hi, 

I bought two Atlas Scientific tri-pump (peristaltic) in order to connect them to a same RPI 4, Tentacle T3 and Mycodo.
I tried to change the I2C address of the 3 pumps of the first tri-pump with mycodo but it return  : 
Command returned: ('error', '2')

### Versions:

 - Mycodo Version: 8.15.8
 - Raspberry Pi Version: RPI 4 2Go
 - Raspbian OS Version: "Raspbian GNU/Linux 11 (bullseye)"

### Reproducibility

1. *[Go to output]*
2. add : Peristaltic pump : Atlas Scientific (I2C)
3. set the default I2C adress (ex: 0X38), bus 1
4. save
5. New i2c adress : 0X28
6. return in log : 

2023-04-17 10:44:37,917 - INFO - mycodo.devices.atlas_scientific_i2c_1_56 - Atlas Scientific Board: PMP, Rev: 2, Firmware: 1.06
2023-04-17 10:44:37,918 - INFO - mycodo.outputs.pump_atlas_ezo_pmp_9ddb0837 - Initialized in 668.0 ms
2023-04-17 10:44:48,994 - INFO - mycodo.outputs.pump_atlas_ezo_pmp_9ddb0837 - I2C Change command: I2C,119
2023-04-17 10:44:49,502 - INFO - mycodo.outputs.pump_atlas_ezo_pmp_9ddb0837 - Command returned: ('error', '2')

How i could change the I2C address of this device ? (I tried with the Ezo : Raspberry-Pi-sample-code too) but same error too ...

Thanks in advance for your advice 

Best reagrds

kizniche, 2023-04-17:
Have you followed the documentation from the manufacturer?

pe2269, 2023-04-17:
Yes i think , the documentation concerning the changement of the I2C address is the same that the single Ezo Pump.

I think it's not a real mycodo issue but an Atlas scientific issue or mine .. :)

[TRI-PMP-BX-Datasheet.pdf](https://github.com/kizniche/Mycodo/files/11250491/TRI-PMP-BX-Datasheet.pdf)

pe2269, 2023-04-17:
A solution was to change the i2c bus for one one the tri-pump !
They keep the same i2c address but not in the same bus ..

pe2269, 2023-04-19:
In the Atlas Scientific documentation of the EZO PUMP (not the TRI-PUMP) p72, it is written that the default protocol lock is 0.
But in the case of the tri pump, Atlas scoezntific engineer answer me : "The protocol lock feature is enabled to prevent accidental address changes.
Disable the lock, change the address, and then re-enable the lock. "

[EZO_PMP_Datasheet.pdf](https://github.com/kizniche/Mycodo/files/11273481/EZO_PMP_Datasheet.pdf)