kizniche/Mycodo #1301: Update MyCodo documentation PDF file? (Currently 8.2.3)

victorhooi, 2023-05-04:
Is there any possibility of having the MyCodo documentation available as a PDF file?

There's an old version (8.2.3) available here - which whilst outdated, is still pretty useful:

https://www.researchgate.net/profile/Kyle-Gabriel/publication/328228021_Mycodo_Environmental_Regulation_System/data/5e30962892851c7f7f08b97f/mycodo-manual-823.pdf

Is there any chance of updating this to the current version please? (Or possibly making it some kind of automated/CI style build).

kizniche, 2023-05-04:
I don't plan to generate PDF manuals again. The online documentation is sufficient.

closed 2023-05-04 by kizniche

victorhooi, 2023-05-04:
Oh bugger =(.

Well, it was more that the PDF file is pretty helpful for ebook readers (e.g. Remarkable tablets, Kindle Scribe etc.)

Either PDF or ePUB would be fine for eink devices.

Was it quite difficult to have the docs output to a PDF version? Is there anything we could do to help?

Or do you think it'd be possible to paste any info on how you did it before, so the community could take a look, mayhaps, or people could try to generate it themselves?

kizniche, 2023-05-04:
FYI the mkdocs pdf plugin doesn't work with the multilingual plugin.