kizniche/Mycodo #1302: Problem installing Highcharts dependency

matiasict, 2023-05-05:
### Describe the problem/bug

I am trying to install the Highcharts dependency, after installing the dependency and restarting backend and frontend, it asks me to install Highcharts again.  It doesn't seem to install.

### Versions:

 - Mycodo Version: *8.15.8*
 - ~Raspberry Pi Version~
 - ~Raspbian OS Version~
 - Intel computer with Ubuntu Server 22.04.2 LTS

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *Go to Setup -> Method*
2. *Click on Method Type and select Time/Data*
3. *Type the name and click on add*
4. *see install highcharts-9.1.2.js dependency - Click on Install All Dependencies*
5. *after installer, restart backend and frontend*
6. *Again go to  Setup -> Method*
7. *Click on Method Type and select Time/Data*
8. *Type the name and click on add*
9. *Again requests to install highcharts-9.1.2.js*

### Expected behavior

*Install dependency and be able to add the Time/Data method*

### Screenshots

[image]

[image]

[image]

### Additional context

1. Installing from the dashboard does not work either
2. Run the commands to see if it solves the problem, but it didn't work.
```
sudo mycodo-commands update-apt
sudo mycodo-commands update-dependencies
sudo mycodo-commands update-pip3
sudo mycodo-commands update-pip3-packages
sudo mycodo-commands update-permissions
sudo mycodo-commands upgrade-master
```
3. I tried deleting the .dependency file, but it didn't work.

AlexValCar, 2023-08-10:
I have the same problem/bug

kizniche, 2023-08-11:
I haven't been able to replicate the behavior, but I do notice you're using different hardware and operating system than the recommended, so it may have something to do with that.

AlexValCar, 2023-08-11:
> I haven't been able to replicate the behavior, but I do notice you're using different hardware and operating system than the recommended, so it may have something to do with that.

Could there be a way to install the dependencies that are giving us problems over ssh?

AlexValCar, 2023-08-12:
> I haven't been able to replicate the behavior, but I do notice you're using different hardware and operating system than the recommended, so it may have something to do with that.

Please help me, please!! i want to intall this dependencys:
 •	highstock-9.1.2.js, highcharts-more-9.1.2.js, data-9.1.2.js, exporting-9.1.2.js, export-data-9.1.2.js, offline-exporting-9.1.2.js and I cant

AlexValCar, 2023-08-18:
> highstock

I solved the problem for you..., your code for wget instructions was wrong...

`wget -nc https://code.highcharts.com/zips/Highcharts-Stock-9.1.2.zip',
                 'unzip Highcharts-Stock-9.1.2.zip -d Highcharts-Stock-9.1.2',
                 'cp Highcharts-Stock-9.1.2/code/highstock.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highstock-9.1.2.js',
                 'cp Highcharts-Stock-9.1.2/code/highstock.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highstock.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/highcharts-more.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highcharts-more-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/highcharts-more.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highcharts-more.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/data.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/data-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/data.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/data.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/exporting.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/exporting-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/exporting.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/exporting.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/export-data.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/export-data-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/export-data.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/export-data.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/offline-exporting.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/offline-exporting-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/offline-exporting.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/offline-exporting.js.map',
                 'sudo rm Highcharts-Stock-9.1.2`

AlexValCar, 2023-08-18:
> I haven't been able to replicate the behavior, but I do notice you're using different hardware and operating system than the recommended, so it may have something to do with that.

I solved the problem for you..., your code for wget instructions was wrong
wget -nc https://code.highcharts.com/zips/Highcharts-Stock-9.1.2.zip',
                 'unzip Highcharts-Stock-9.1.2.zip -d Highcharts-Stock-9.1.2',
                 'cp Highcharts-Stock-9.1.2/code/highstock.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highstock-9.1.2.js',
                 'cp Highcharts-Stock-9.1.2/code/highstock.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highstock.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/highcharts-more.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highcharts-more-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/highcharts-more.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/highcharts-more.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/data.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/data-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/data.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/data.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/exporting.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/exporting-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/exporting.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/exporting.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/export-data.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/export-data-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/export-data.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/export-data.js.map',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/offline-exporting.js /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/offline-exporting-9.1.2.js',
                 'sudo cp Highcharts-Stock-9.1.2/code/modules/offline-exporting.js.map /var/mycodo-root/mycodo/mycodo_flask/static/js/user_js/offline-exporting.js.map',
                 'sudo rm Highcharts-Stock-9.1.2

kizniche, 2023-08-18:
The command works on the recommended hardware and operating system. Please identify the issue rather than merely suggesting something be changed.

zacharyneveu, 2023-10-18:
Ran into this on a fresh Pi 3 with Raspbian. Not sure of the cause (looked at the code and none of the shell commands seem problematic), but manually copying the files resolved the issue. Maybe a permissions issue of some kind?

kizniche, 2023-10-18:
I looked at this issue again and was able to replicate it. As such, I was able to come up with a fix. The release of [v8.15.11](https://github.com/kizniche/Mycodo/releases/tag/v8.15.11) should resolve this issue.

closed 2023-10-18 by kizniche