kizniche/Mycodo #1303: No data under live measurements

Elfrederiko, 2023-05-08:
Hi 
My codo 8.15.8

No more data under live measurements I think this happend after an update of Mycodo.
At first I thought it was an hardware issue with one or more sensors,  But then I checked with basic Raspberry CPU/GPU temp and i got the same issue.  
With log level Debug enabled: I can see the it reeds the temp in Mycodo logs
[image]
But nothing under live measurements:
[image]

Someone any ideas ?

kizniche, 2023-08-11:
Likely the measurement database configuration misconfigured. What are the settings currently set to in General Configuration and what version database are you using?

ebeFuLan, 2023-08-27:
Raspberry Pi OS Lite (Bullseye, Kernal Version 6.1)
Mycodo: 8.15.9

I'm having this same issue. The logs are showing measurements being taken from the sensor, but the Live Data screen reports "No Data".
The Time Series Database settings (general configuration page) are:
Database: Influxdb 2.x (I've tried w/ 1.x but get the same issue)
Port: 8086
Database Name: mycodo_db
Database Username: mycodo

I've also tried adding a custom measuremnt with the measurement ID found in the input configuration. This was unsuccessful as well.

The mycodo.log file is showing the sensor taking measurements:

2023-08-27 14:28:12,689 - DEBUG - mycodo.controllers.controller_input_a658d7a0 - Adding measurements to InfluxDB with ID a658d7a0-e635-48d0-9417-044e684df27f: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.43911814952735, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 268785)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 64.42115056287275, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 268914)}, 2: {'measurement': 'pressure', 'unit': 'Pa', 'value': 101363.77573, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 429095)}, 3: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.339725345600346, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 429409)}, 4: {'measurement': 'altitude', 'unit': 'm', 'value': -3.228, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 429639)}, 5: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1026.19903996614, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 12, 429868)}}
2023-08-27 14:28:27,679 - DEBUG - mycodo.controllers.controller_input_a658d7a0 - Adding measurements to InfluxDB with ID a658d7a0-e635-48d0-9417-044e684df27f: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.444216495618456, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 259341)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 64.4655252798372, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 259471)}, 2: {'measurement': 'pressure', 'unit': 'Pa', 'value': 101359.18535, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 416207)}, 3: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.355377674082966, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 416522)}, 4: {'measurement': 'altitude', 'unit': 'm', 'value': -2.846, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 416749)}, 5: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1025.2341631138322, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 27, 416974)}}
2023-08-27 14:28:43,083 - DEBUG - mycodo.controllers.controller_input_a658d7a0 - Adding measurements to InfluxDB with ID a658d7a0-e635-48d0-9417-044e684df27f: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.444216495618456, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 306793)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 64.45450455546599, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 306927)}, 2: {'measurement': 'pressure', 'unit': 'Pa', 'value': 101370.06186, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 538621)}, 3: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.352690591852152, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 538932)}, 4: {'measurement': 'altitude', 'unit': 'm', 'value': -3.751, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 539156)}, 5: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1025.5521310369998, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 42, 539382)}}
2023-08-27 14:28:57,687 - DEBUG - mycodo.controllers.controller_input_a658d7a0 - Adding measurements to InfluxDB with ID a658d7a0-e635-48d0-9417-044e684df27f: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 23.444216495618456, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 263894)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 64.46001492216155, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 264024)}, 2: {'measurement': 'pressure', 'unit': 'Pa', 'value': 101364.6236, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 420705)}, 3: {'measurement': 'dewpoint', 'unit': 'C', 'value': 16.35403418453669, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 421017)}, 4: {'measurement': 'altitude', 'unit': 'm', 'value': -3.298, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 421244)}, 5: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1025.3931469452955, 'timestamp_utc': datetime.datetime(2023, 8, 27, 21, 28, 57, 421469)}}

Do you have any ideas on how to get it to show on the "Live Data" page?

kizniche, 2023-08-27:
Likely duplicate of #1292

kizniche, 2023-08-27:
It is only likely a similar issue if you had a working system and only after an upgrade did it display this behavior.

ebeFuLan, 2023-08-27:
No it didn't. I'm experiencing this with a new install. Also, this is the only sensor (BME 280, library RPi.bme280) that has been added.

kizniche, 2023-08-28:
Then it's likely a separate issue entirely and shouldn't be discussed here. If it's a Pi zero, you're likely experiencing #1292, which isn't a Mycodo issue at all.

Elfrederiko, 2023-09-26:
> It is only likely a similar issue if you had a working system and only after an upgrade did it display this behavior. Did your behavior only occur after an upgrade?

> Likely the measurement database configuration misconfigured. What are the settings currently set to in General Configuration and what version database are you using?

[image]