kizniche/Mycodo #1304: Energy Usage (Outputs) and Energy Usage (Amp Measure)) Menu Options - Produce, "Error 500" - Verify flux-enabled=true in the [http] section of the InfluxDB config

smguth, 2023-05-10:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug
[image]
(Energy Usage (Outputs) and Energy Usage (Amp Measure)) Menu Options - Produce, "Error 500"

`Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. However, first search through the [github issues](https://github.com/kizniche/Mycodo/issues) and the [forum](https://forum.radicaldiy.com/search) to see if this issue has already been reported. If it hasn't, submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.15.8
Database: 16b28ef31b5b
Model: Raspberry Pi 4 Model B Rev 1.1

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 2528, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/USER/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1057, in page_energy_usage_outputs
    dict_outputs, misc, output, OutputChannel, custom_options_values_output_channels)
  File "/home/USER/Mycodo/mycodo/utils/tools.py", line 362, in return_output_usage
    each_output.unique_id, 86400, output_channel=each_channel.channel) / 3600
  File "/home/USER/Mycodo/mycodo/utils/influx.py", line 498, in output_sec_on
    past_sec=past_seconds)
  File "/home/USER/Mycodo/mycodo/utils/influx.py", line 300, in query_string
    limit=limit)
  File "/home/USER/Mycodo/mycodo/utils/influx.py", line 281, in query_flux
    tables = client.query_api().query(query)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/client/query_api.py", line 204, in query
    async_req=False, _preload_content=False, _return_http_data_only=False)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/service/query_service.py", line 279, in post_query
    (data) = self.post_query_with_http_info(**kwargs)  # noqa: E501
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/service/query_service.py", line 320, in post_query_with_http_info
    urlopen_kw=kwargs.get('urlopen_kw', None))
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/_sync/api_client.py", line 348, in call_api
    _preload_content, _request_timeout, urlopen_kw)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/_sync/api_client.py", line 177, in __call_api
    _request_timeout=_request_timeout, **urlopen_kw)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/_sync/api_client.py", line 395, in request
    **urlopen_kw)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/_sync/rest.py", line 318, in POST
    **urlopen_kw)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb_client/_sync/rest.py", line 261, in request
    raise ApiException(http_resp=r)
influxdb_client.rest.ApiException: (403)
Reason: Forbidden
HTTP response headers: HTTPHeaderDict({'Content-Type': 'text/plain; charset=utf-8', 'Request-Id': '8ef6fbc8-eee8-11ed-80fb-dca63248a74b', 'X-Content-Type-Options': 'nosniff', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': '1.8.10', 'X-Request-Id': '8ef6fbc8-eee8-11ed-80fb-dca63248a74b', 'Date': 'Wed, 10 May 2023 04:10:11 GMT', 'Content-Length': '100'})
HTTP response body: b'Flux query service disabled. Verify flux-enabled=true in the [http] section of the InfluxDB config.\n'
`

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *[e.g. 3B+]*
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Log into Mycodo
2. From the Menu Options select "More" > (Energy Usage (Outputs) || Energy Usage (Amps Measure))

### Expected behavior
 The expected behavior would be a non Error 500 page.

### Screenshots

[image]
### Additional context

So I think this here is my issue but I'm not sure how to edit this config.
Verify flux-enabled=true in the [http] section of the InfluxDB config

I am bush league at best with the InfluxDB so forgive my ignorance here.  I was poking around the user-interface hoping there was a enable Flux button but I didn't see it. Hoping this isn't something I've hosed up.

smguth, 2023-05-10:
Just noticed my clock is off too... says its tomorrow...

kizniche, 2023-07-11:
Flux is enabled upon install. See:

https://github.com/kizniche/Mycodo/blob/73a324371d39a39c081755f3f88420fe930d42b4/install/setup.sh#L189-L194

and:

https://github.com/kizniche/Mycodo/blob/73a324371d39a39c081755f3f88420fe930d42b4/mycodo/scripts/upgrade_commands.sh#L436

For Flux not to be enabled, there could have been an issue during install.