kizniche/Mycodo #1314: Function: Backup to Remote Host (rsync) - not working with influxdb v2

Mondgans, 2023-06-14:
After a hardware failure I installed a new mycodo system with an influxdb version 2.
When creating a backup function I get the following error:
```
2023-06-14 09:48:13,093 - INFO - mycodo.controllers.controller_function_a3c30bde - Activated in 197.5 ms
2023-06-14 09:53:14,636 - ERROR - mycodo.controllers.controller_function_a3c30bde - Exception while running loop()
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 79, in loop
    self.run_function.loop()
  File "/home/falko/Mycodo/mycodo/functions/backup_rsync.py", line 291, in loop
    self.backup_measurements()
  File "/home/falko/Mycodo/mycodo/functions/backup_rsync.py", line 360, in backup_measurements
    status, saved_path = create_measurements_export(save_path=path_save)
TypeError: create_measurements_export() missing 1 required positional argument: 'influxdb_version'
```
rsync is working. The 'backup_settings' directory is created at the remote host but stays empty.
Regards Falko

kizniche, 2023-06-15:
Thanks for the bug report. I'll look into this.

kizniche, 2023-06-16:
I just committed a potential fix. You can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) if you would like to test the latest code before release. If you do, please let me know if the fix worked.

Mondgans, 2023-06-18:
The behavior has changed. But I can not interpret this:
`2023-06-18 14:36:31,601 - INFO - mycodo.measurement_db - Influxdb info: {'db_name': 'influxdb', 'db_version': '2', 'influxdb_installed': True, 'influxdb_retention_policy': 'infinite', 'influxdb_version': '2.6.1', 'influxdb_host': 'localhost', 'influxdb_port': '8086'}
2023-06-18 14:36:34,116 - INFO - mycodo.tools - out: b'', error: b'2023/06/18 14:36:32 INFO: Downloading metadata snapshot\n2023/06/18 14:36:32 INFO: Backing up TSM for shard 1\n2023/06/18 14:36:32 INFO: Backing up TSM for shard 2\n', status: 0`

kizniche, 2023-07-11:
You likely have it misconfigured. This issue appears to be solved with the fix.

closed 2023-08-21 by kizniche