kizniche/Mycodo #1316: Please add 2-way MQTT communication for on/off MQTT Publish Output.

LucidEye, 2023-06-17:
I recently built a WiFI irrigation controller using an ESP32 attached to a 4-channel relay board which receives on/off commands from Mycodo via MQTT.
The system has been working ok so far, except today I noticed that one of the valves was open after it's Mycodo timer function should have shut it off.  There were no indicators in Mycodo to show that the relay was still turned on... all the mycodo indicators showed the relay in an off state.
The obvious reason for this problem is that for some reason the ESP32 did not recieve the "off" command from Mycodo... either due to a network problem, or an MQTT problem.
Because there is currently no 2-way communication between Mycodo and the remote MQTT device built into the Output module, there is no way to monitor the remote device to see if it received and acknowledged the MQTT packet.
Essentially Mycodo is operating "in the blind" when it comes to the MQTT output.  It just sends the on/off command anf assumes that the remote device received and performed the command.

It would be beneficial to have the Mycodo MQTT Output be capable of not only sending MQTT commands to the remote device, but also receive status updates **from** the device. ESP firmware like Tasmota and ESPHome have MQTT device-status topics that they publish specifically for the purpose of informing automation software of their current state.  This would offer something closer to a "closed loop" system and help prevent situations where a lapse in WiFI signal might cause a situation like described above where an output is showing one state in Mycodo, but in actuality is in the opposite state.  I know this still isn't a true closed loop system, but it would at least offer some extra safety and peace of mind for remote switching applications.

I know that the Mycodo and Tasmota APIs should also be capable of 2-way communication with each other as well, but I have not had time to explore this direction yet, so I don't know if the better solution would be to do this with the API, or via MQTT.

Essentially the idea would be to have remote devices running firmware like Tasmota to be able to report their current status to Mycodo, so that Mycodo can tell what the device state actually is rather than assuming. Similar to how the Kasa plugs and power strips are capable of status reporting to Mycodo (if you manually turn on a Kasa plug using it's physical power button, Mycodo will "know" the device was turned on).
So I'm not sure if it would be possible/feasible to add another "Topic" field to the MQTT on/off Output that would be for receiving MQTT status updates.  
[image]

kizniche, 2023-06-19:
Why not program the device to receive a duration to turn the relay on, rather than relying on an inherently dangerous need to also receive an off command following the on command?

LucidEye, 2023-06-19:
Not sure yet if that is possible with Tasmota....  There are actually timer functions that can be set in Tasmota, and i'm looking into how they work, but I've been very busy so it's slow-going at the moment :-(
There is a Tasmota command called "backlog" that allows you to send several commands at once to a Tasmota device.... so that might be a way to make it work.
Also, the current MQTT On/Off Function is only setup to send on and off commands.... not sure I can append a "timer" command to the "on" command with the current Mycodo function module, wouldn't it need to be reconfigured to send a JSON packet instead of a single on or off command? 
I'm going to try setting up my test system to see if the MQTT on/off function will send the backlog command so I can append several commands at once, something like  BackLog, power1=on, timer=300
The problem with this is that if Mycodo is not the thing keeping track of the time that output is on, then Mycodo will show that output as off, right? which means I can not graph that data, nor can I see in real-time what state the outputs are in. So we are back to the same problem of needing a 2-way communication so that Mycodo can read the actual current state of the remote device.

kizniche, 2023-06-19:
>if Mycodo is not the thing keeping track of the time that output is on
Why wouldn't it be? All duration outputs record the time sent to them.

LucidEye, 2023-06-19:
But how would I send an on+timer command, and then not send an off command?  
Or do I just have Mycodo send the off command into null space or something.... so both Mycodo and Tasmota are both running timers?

kizniche, 2023-07-29:
This would be a good candidate for a new Tasmota MQTT Output, not modifying the current MQTT Output. How does one get the current state of the output?

LucidEye, 2023-07-29:
The MQTT topic format Tasmota uses to report it's state is:
tele/device/<topic-name>/STATE

To send commands to a Tasmota device the MQTT format is:
cmnd/device/<topic-name>/<command-payload>