kizniche/Mycodo #1318: Problems with Ramping Duty Cycle Action and fixed Duty Cycle Actions "looping" in Trigger: Output On/Off Function

LucidEye, 2023-06-28:
- Mycodo Version:8.15.8
 - Raspberry Pi Version: 4B 2GB
 - Raspbian OS Version: Bullseye

When setting up a Ramp Duty Cycle Action inside a Trigger: Output On/Off, the expected behavior would be that the duty cycle starts at the selected Duty Cycle: Start percent, and then ramps up or down to the selected Duty Cycle: End percent, and then **stays** at the Duty Cycle: End percent indefinitely until another Action or Function tells it to do so.
[image]

This should also be the same behavior if I have set multiple individual fixed duty cycles separated by Pause Action timers.... the Actions should start at the top of the list, and then end at the last action on the list, and the duty cycle should stay set at whatever that last action is set for.
[image]

What is happening instead is that after reaching the Duty Cycle: End or the last action on the list, it is repeatedly "looping back" to the Duty Cycle: Start percent, or the first Action on the list.
The Ramp Duty Cycle will get stuck in this "loop" and not shut off when it is supposed to... as in when the function is deactivated, it is still running out the remainder of this "loop" and doesn't actually deactivate until the loop is finished.
I don't believe this is the correct behavior, at least it did not used to behave this way until the last Mycodo update.

kizniche, 2023-07-11:
Just tested and the Ramp Duty Cycle Action worked as expected, remaining on the end duty cycle at the end.

kizniche, 2023-07-11:
I just tested both action configurations and they work as expected. You're complicating the matter by not using an Execute Actions Function to test.

LucidEye, 2023-07-11:
"You're complicating the matter by not using an Execute Actions Function to test."
I'm not sure what you mean by that.  I have used both the previously mentioned configurations and have never seen this behavior until after upgrading to the 64 bit setup, so why would I need to use another function to test this? What will it tell me? 
What might be causing this, why do I seem to be having all of these issues after moving to a 64-bit system?
Are there dependencies or other things in the settings that don't import well between 32 and 64 bit systems?
Many of my settings and functions were created several versions ago and have been imported several times... Is there possibly some way this could be corrupting the system somehow?  It just doesn't make any sense having done Mycodo setups so many times now with none of these issues.  
I

kizniche, 2023-07-11:
Don't know what to tell you other than I can't replicate what you're describing. If I can't replicate it, it's impossible for me to do anything. You need to provide a detailed step-by-step list that replicates the issue.