kizniche/Mycodo #1321: Working w/ Banana PI & Mycodo Security Issues?

FlooSz, 2023-07-12:
Hi, first of all _thank you_ @kizniche for the great software and the constant further development! 

I discovered some security gaps that could become dangerous if you use DynDNS instead of VPN, for example. Because I was wondering why the connection to Mycodo is displayed as insecure in my browser, I delved into the depths of the code to understand exactly how Mycodo works and I noticed that on the one hand in the API, e.g. verify is set to False and that does not use SSL encryption. I was also wondering why implement TSL/SSL if you don't use it by default, or do you get an error when you start Flask with the --SSL command because you only have a test certificate and it doesn't work any other way? Or is that unintentional? Snyk.io shows quite a lot of errors, mainly in Docker, but it seems to me that there could also be errors in Mycodo itself e.g. https://learn.snyk.io/lessons/directory-traversal/python/?loc=fix- pr probably just have a look for yourself, you can do hundreds of tests for free every month, and so I can keep the text here much shorter than if I post hundreds of errors here now, there should be fewer errors when going from nginx:1.17.6 to nginx :1.23.3, but I'm sure you can rate it all much better what will works with Mycodo. 

I had some problems with the Banana Pi at the beginning with the installation, so that I could complete this I had to install some dependencies in the virtenv, but I still have problems with some dependencies like Highstock, Highcharts, hope the sensors work then everything, I don't have any yet, but I'm confident that I'll just have to map the GPIO pin or something. Some log files are empty for me. I had to create them all by hand and adjust the permissions, but haven't managed to see all the logs yet. E.g. Mycodo Deamon Keepup, Is that how it should be or when should something be inside? It was somehow clear to me that the Raspberry Pi side does not deliver any correct data. (I've tried everything you said in the other threads.)

kizniche, 2023-07-12:
If this is only docker related, there's an issue already for that discussion. In the [docker readme](https://github.com/kizniche/Mycodo/blob/master/DOCKER.md), it's explained docker is experimental:

>This is currently experimental
>
>Please do not submit github issues for Docker-related problems. Also do not expect this feature to remain consistent (i.e. previous builds may not be compatible with future builds). Join https://github.com/kizniche/Mycodo/issues/637 for Mycodo Docker discussion.

For your other comments:

>the API, e.g. verify is set to False and that does not use SSL encryption

SSL is forced by Flask_Talisman unless you disable SSL in the General Configuration

>I was also wondering why implement TSL/SSL if you don't use it by default

I'm not sure what you mean, because SSL is forced by default

FlooSz, 2023-07-12:
> Wenn dies nur mit Docker zu tun hat, gibt es bereits ein Problem für diese Diskussion. In der [Docker-Readme-Datei](https://github.com/kizniche/Mycodo/blob/master/DOCKER.md) wird erklärt, dass Docker experimentell ist:
> 
> > Dies ist derzeit experimentell
> > Bitte reichen Sie Github-Probleme nicht für Docker-bezogene Probleme ein. Erwarten Sie auch nicht, dass diese Funktion konsistent bleibt (dh frühere Builds sind möglicherweise nicht mit zukünftigen Builds kompatibel). Nehmen Sie an der Mycodo Docker-Diskussion [Nr. 637 teil.](https://github.com/kizniche/Mycodo/issues/637)
> 
> Für Ihre anderen Kommentare:
> 
> > Die API, z. B. „Verify“, ist auf „False“ gesetzt und verwendet keine SSL-Verschlüsselung
> 
> SSL wird von Flask_Talisman erzwungen, es sei denn, Sie deaktivieren SSL in der allgemeinen Konfiguration
> 
> > Ich habe mich auch gefragt, warum man TSL/SSL implementieren sollte, wenn man es nicht standardmäßig verwendet
> 
> Ich bin mir nicht sicher, was Sie meinen, da SSL standardmäßig erzwungen wird

FlooSz, 2023-07-12:
No im meaning when I connect from my PC to the Banana Pi, theres a certificate error is displayed and you have to click on advanced and visit anyway, and the https is crossed out

kizniche, 2023-07-12:
There's a self-signed certificate generated during install. You can always replace that with your own certificate if you want a signing authority to verify it. Even sites with self-signed certificates are encrypted. The HTTPS being crossed out doesn't mean it's unencrypted, it only means the self-signed certificate couldn't be verified (which is expected).

closed 2023-08-11 by kizniche