kizniche/Mycodo #1326: Output: On/Off MQTT Publish does not work with Trigger Output: On/Off Function

LucidEye, 2023-07-28:
Raspberry Pi Model 2
Raspberry Pi OS Lite 32-bit version
Mycodo 8.15.8

The Situation:

I have Mycodo controlling a 4-channel remote (wifi) relay via Output On/Off: MQTT Publish (paho-mqtt).
The remote relay is physically connected to an ESP32 running Tasmota.
Manually switching the relays on and off from the Mycodo Outputs page works just fine. 
Turning relays 1-4 on and off using timer Functions works just fine.
These relays (1-4) operate my irrigation zone valves.
However, I also need to activate a 5th, separate, single relay in tandem with those other 4 relays... so any time relays 1-4 are activated, I also need relay 5 to turn on, and likewise be off whenever relays 1-4 are off.
That 5th relay is what turns on the 24VAC transformer to power the valve solenoids... due to excessive heat, I need to keep it off until it is actually needed.
The easiest way to achieve this using Mycodo is to setup a series of Trigger: Output On/Off functions that "watch" relays 1-4, and activate & deactivate relay 5 depending on the state of relays 1-4.
[image]
[image]
[image]

(Ideally, Tasmota and the ESP32 should handle this simple switching task, but until I figure out how to program this using Tasmota Rules or Berry Script, I need Mycodo to do it or none of our plants get watered.... this may be a while as I have been dealing with a serious shoulder injury and may yet need surgery).

The Problem:

As I said, I can activate the irrigation zones 1-4 manually from the Outputs page and with automation using Mycodo Functions.
[image]

However I ran into a problem when leaving it up to Mycodo to handle the automation.
What is happening (or not happening) is that when I leave the irrigation zone timer Functions activated, and they turn on the zone valve relays (1-4), for some reason the 5th relay is not being triggered by the Trigger: Output On/Off functions I setup specifically for this purpose.
The weird thing is that this all works fine if I manually activate the zone relays 1-4 directly from the Outputs page... when I activate/deactivate relays 1-4, the 5th relay also automatically activates/deactivates.
However when the Irrigation timer Functions activate relays 1-4, the 5th relay is not being activated.  
I can confirm this by clicking on the 'execute actions' button in the Functions and I can see that the irrigation valve relays (1-4) are being activated, but for some reason the Trigger: Output On/Off functions are not activating relay 5?
I can also confirm this by watching my MQTT broker using MQTT Explorer... it never receives the on or off payload from Mycodo when the irrigation timer functions turn on the valve relays 1-4.
But it does work when manually activating the relays from the outputs page.
The expected behavior is that any time the Mycodo Outputs for relays 1-4 are activated/deactivated, the Output for relay 5 should also be activated/deactivated, regardless of what Mycodo method or function was used to trigger relay Outputs 1-4.

kizniche, 2023-07-28:
Your If State should be "ON (no or any duration)"

LucidEye, 2023-07-29:
Ok... can you please explain that... because to my thinking it should work if it's on, regardless of any duration.... they should almost activate simultaneously considering how fast the Pi is computing all of this.

kizniche, 2023-07-29:
"no or any duration" would trigger if the output is turned on (without a duration) or turned on for a duration.

kizniche, 2023-07-29:
Yoy could also select "On (any duration)", but selecting "On (no duration)" simply won't work.

LucidEye, 2023-07-29:
Fantastic!  Confirmed and tested... It's now actually opening the valves and getting water to the plants once again.
So this nasty summer heat (111F) has not been nice to some of the electronics I had in the irrigation control box I built last month.  I ended up burning out 2 voltage regulators, which took out 2 ESP32s and a 4 channel relay board, as well as the 24VAC transformer burning out too :-(
I've since redesigned everything and got some shade to better protect the box, but I had to add that 5th relay to keep the big transformer powered down until it was actually needed to reduce the heat inside the control box.
So why does selecting On (no duration) not work?  If A is on, it's on, and if A is ever on, B should also be on as soon as A is on.  The Off state doesn't have these conditions.  It's a bit confusing, especially when it means the difference between the whole system working or not.

kizniche, 2023-07-29:
There's a difference between turning an output on (no duration) and turning an output on for a duration. There are many use cases where you would want to distinguish these types of events.

kizniche, 2023-07-29:
Your outputs are turning on for a duration, therefore if you select "On (no duration)", the condition "no duration" is not met.

LucidEye, 2023-07-29:
Of course, but it still doesn't make sense to me as to why On (no duration) doesn't work... if on is on.

kizniche, 2023-07-29:
On can be for a duration or without a duration.

LucidEye, 2023-07-29:
Right, but I am controlling the duration with another Function entirely.
Is that why?

kizniche, 2023-07-29:
It doesn't matter what is controlling it, only that there is a duration, therefore the truth of "no duration" is False, therefore the action will not execute.

LucidEye, 2023-07-29:
Ok... so even if another function is setting a duration for the output to run (which is what my irrigation valve timer functions are doing)... in that specific case, I will need to use the On (any duration) setting for a Trigger Function.
But in the case where the function activates an output for an open-ended time (no specific duration defined) then I can use the On (no duration) setting?

kizniche, 2023-07-29:
Yes. And if you wanted either to trigger the action, select what I initially recommended, "On (no or any duration)".

LucidEye, 2023-07-29:
OK.. I guess that makes sense now.
Thank you once again for your help and for clarifying those settings.
Glad it was just a setting and not a bug :-P

kizniche, 2023-07-29:
You're welcome. Always nice to fix something just by chatting.

closed 2023-07-29 by kizniche

kizniche, 2023-07-29:
Oof