kizniche/Mycodo #1328: libcamera livestream on mycodo

Abrio97, 2023-08-04:
I've connected my Raspberry Pi camera module 3 to my raspberry pi and added it into mycodo as a libcamera, but I can't create a livestream for my dashboard. 
I tried to add my module 3 camera as a picamera or opencv but it didn't work either. I assume that wouldn't work anyway since it's a libcamera.
Can someone help me out on how to create a livestream with my camera module 3 on mycodo?

kizniche, 2023-08-05:
Have you been able to acquire an image (by any means) to verify you have the hardware connected and the Linux software set up properly?

LucidEye, 2023-08-11:
I seriously doubt that the V3 camera, which is 12 megapixels, is going to give you a stable live video stream.... it's simply trying to process too much data.  Even a V1 camera, which is only 5 megapixels, has a hard time streaming live video in Mycodo at anything over 1-2 FPS.  The first thing I would try is setting the camera video resolution WAY down to something like 800x400 and then test if you can stream video.  Also keep in mind that the libcamera library is still under development, so it may not work as well as the older legacy libraries yet.

closed 2024-10-04 by kizniche