kizniche/Mycodo #1329: When a conditional controller disables a conditional controller, its status is not refreshed in the function view

lalebarde, 2023-08-05:
Hi Kyle,

When a conditional controller disables a conditional controller, its status is not refreshed in the function view. If I refresh the browser tab, then the status is updated: function frame becomes turns off from green to red.

 - Mycodo Version: 8.15.8
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

Reproducibility : always, also in previous versions

kizniche, 2023-08-05:
This is expected behavior, as the functionality never existed for Functions. It also doesn't exist for any controller that can be disabled (Inputs, Cameras, etc.).

lalebarde, 2023-08-05:
Thanks Kyle, it would be nice to have it. In tuning phase, it takes time to update the browser view.