kizniche/Mycodo #1332: Enable Private Vulnerability Reporting in GitHub

ZuhairORZaki, 2023-08-16:
In your repository, we have found a bug that may require your attention. We do not want to disclose the details. Therefore, we request you to enable private vulnerability reporting in your repository.

### Sponsorship and Support:
This work is done by the security researchers from OpenRefactory and is supported by the [Open Source Security Foundation (OpenSSF)](https://openssf.org/): [Project Alpha-Omega](https://alpha-omega.dev/). Alpha-Omega is a project partnering with open source software project maintainers to systematically find new, as-yet-undiscovered vulnerabilities in open source code - and get them fixed - to improve global software supply chain security.

The bug is found by running the iCR tool by [OpenRefactory, Inc.](https://openrefactory.com/) and then manually triaging the results.

kizniche, 2023-08-17:
Thank you. I've enabled private vulnerability reporting.

kizniche, 2023-10-03:
Published at https://github.com/kizniche/Mycodo/security/advisories/GHSA-jjhj-w3hx-5gqp

closed 2023-10-03 by kizniche