kizniche/Mycodo #1335: Activating AnyLeaf PH sensor fails.

ghunti, 2023-09-18:
### Describe the problem/bug

Activating [AnyLeaf PH sensor](https://www.anyleaf.org/ph-module) fails. 

### Versions:

 - Mycodo Version: 8.15.9
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: PI OS 64-bit. Debian Bullseye with Raspberry Pi Desktop

### Reproducibility

1. Go to: Setup -> Input
2. Add the "AnyLeaf: AnyLeaf pH: Ion concentration (anyleaf) [I2C]" input
3. Install all the dependencies
4. Go back to: Setup -> Input
5. Activate the AnyLeaf input

### Expected behavior

The AnyLeaf sensor should be active and provide measures.
Although the AnyLeaf input displays as "active" on the interface, the logs show an error, which I think is preventing the AnyLeaf sensor from functioning properly and displaying any measures.

### Screenshots
The error log:
[image]

The AnyLeaf input configuration:
[image]

### Additional context
To see if the sensor is working, I've run the [anyleaf-python examples](https://github.com/anyleaf/anyleaf-python), and everything seems to be working:
[image]

Thank you :-)

kizniche, 2023-09-18:
Are you setting the correct bus and address? You don't provide a copy of the I2C device tree to determine if you are.

kizniche, 2023-09-18:
What is the output from `sudo i2c-detect -y 1` ?

kizniche, 2023-09-19:
Looking closer at the trace back in your screenshot, it looks like the issue is with the ads1115 library, or rather, the call of a function of that library. It's been a while since the anyleaf library version has been updated, so this had likely been fixed. I'll check their library changelog.

kizniche, 2023-09-19:
Looks like the library is still at 0.9.1 https://pypi.org/project/anyleaf/

kizniche, 2023-12-19:
@David-OConnor do you know if this is a known issue?