kizniche/Mycodo #1336: [Action] Equation (Single-Measurement) is not working for MQTT Subscribe (value payload) Input

LucidEye, 2023-09-23:
Mycodo 8.15.9
R-Pi 4B 4GB
Latest Bullseye R-Pi OS

I have an MQTT Subscribe (value payload) Input setup.  The input is working, receiving telemetry from the remote sensor via MQTT.
However, this sensor is a little off, and I am using the Equation (Single-Measurement) Action to "adjust / calibrate" the measurement.
The problem is that this action does not appear to be working.

Here is a shot of the sensor's direct reading from the ESP32 firmware running the remote sensor...
[image]

Here are the [Action] Equation (Single-Measurement) settings...
[image]

Here are the sensor readings from the Mycodo Live Measurements page...
[image]

As you can see, the measurements still read the same after being received by the Mycodo input, and the Equation (single-measurement) doesn't appear to have been applied.

I am using the Equation (single-measurement) on another sensor that is on I2C, and that is working, it is adjusting the measurement on that input, just not the MQTT input.

kizniche, 2023-11-21:
I found the issue, which is related to any Input that has a variable number of measurements. These inputs store measurements to the database differently than others and, consequently, are not executing actions. I'm currently attempting to develop a solution.

kizniche, 2023-11-21:
There's now a yet-to-be released fix on the master branch that should fix this.

LucidEye, 2023-11-21:
Thank you.  I would prefer to wait until the next official release to test this on my live system since I currently don't have a dedicated test system running anymore... I needed that Pi for another project.

closed 2024-10-04 by kizniche