kizniche/Mycodo #1340: GNU License Window Unresponsive

neutralgenius, 2023-10-14:
### Describe the problem/bug

Upon execution of "curl -L https://kizniche.github.io/Mycodo/install | bash" the installation downloads and launches, but hangs up on the portion where you can accept the GNU license. You cannot select "Yes" or "No" through any input key command: ex. "Y", "ENTER", "YES+ENTER", "CONFIRM", or even adding "echo y |" in front of the command did not work. The installation is now on the device, but the installation process is unexecuted.

### Versions:

 - Mycodo Version: Latest version: 8.15.9
 - Raspberry Pi Version: Using Raspberry Pi 3B
 - Raspbian OS Version: Raspberry Pi OS with desktop and recommended software

    Release date: October 10th 2023
    System: 64-bit
    Kernel version: 6.1
    Debian version: 12 (bookworm)
    Size: 2,725MB

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Flash Debian version onto SD card.
2. Boot up new OS
3. Connect to Network
4. Launch Command window
5. Input "curl -L https://kizniche.github.io/Mycodo/install | bash" and Enter

### Expected behavior

It downloads the repository and installs Mycodo

### Additional context

I was able to circumnavigate the issue by going directly to the install file, "setup.sh", now in the file directory using the right click text-editor option and copying "sudo /bin/bash ~/Mycodo/install/setup.sh" from it.

I opened an terminal window and executed "sudo /bin/bash ~/Mycodo/install/setup.sh". The GNU license screen pops up and a simple "ENTER" key selects the highlighted response. The installation continues as expected.

EthanDarling8, 2023-10-15:
Same thing is happening for me. I have to exit the installer with `ctrl+shift+c` then navigate to `/Mycodo/install` and run `sudo ./setup.sh`. Then it failed when trying to connect to localhost so I had to install nginx manually and then it worked.

kizniche, 2023-10-15:
This issue has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/mycodo-installation-issues-could-not-connect-to-localhost/1637/10

EthanDarling8, 2023-10-15:
> This issue has been mentioned on **Radical DIY Forum**. There might be relevant details there:
> 
> https://forum.radicaldiy.com/t/mycodo-installation-issues-could-not-connect-to-localhost/1637/10

I can confirm that it seems to be an issue with the latest debian (12). Grabbing a Bullseye version of pi os lite worked.

kizniche, 2023-10-15:
Should be fixed with (8.15.10](https://github.com/kizniche/Mycodo/releases/tag/v8.15.10).

closed 2023-10-15 by kizniche

neutralgenius, 2023-10-16:
Got it working! Went to Debian 11 with a re-flashed SD, enabled HTTP on Chrome, and installed Virtualenv. Thanks for the quick input, I'm a big fan of this project!

kizniche, 2023-10-16:
The 8.15.10 release was after I verified all bugs were fixed with the latest Raspberry Pi OS (Debian 12) and tested a full install, so it isn't necessary to use the old Debian 11 anymore.