kizniche/Mycodo #1341: Failed to restart nginx.service

mfittipaldi90, 2023-10-16:
### Describe the problem/bug

When installing Mycodo 8.15.10, cannot connect to localhost, and cannot create an admin user to log in.

### Versions:

 - Mycodo Version: 8.15.10
 - Raspberry Pi Version: 4
 - Raspbian OS Version: 12 Bookworm

### Reproducibility

1. run "sudo /bin/bash ~/Mycodo/install/setup.sh" in terminal
2. accept license agreement
3. select en language and continue installation
4. select "yes" to install database and wait for installation

### Expected behavior
Expect install to install Mycodo and influxdb

### Additional context
See setup.log attached

[setup.log](https://github.com/kizniche/Mycodo/files/12911000/setup.log)

kizniche, 2023-10-16:
Are you sure you're running 8.15.10? Your setup log doesn't indicate this, and you aren't installing using the install script in the install instructions.

mfittipaldi90, 2023-10-16:
Very new at this. I tried running the install as explained in the wiki but
I had a license page issue that was not responding and found a workaround
in another post.

The files were downloaded using “
curl -L https://kizniche.github.io/Mycodo/install | bash”
 Command. Does it download the latest version?

kizniche, 2023-10-16:
Your "1." under Reproducibility indicates you the setup.sh not install.
Did you delete all previous files before issuing the install command?

kizniche, 2023-10-16:
Oh, I see an issue with the install script. I need to build the docs.

kizniche, 2023-10-16:
Docs have finished building, so the updated install script is now live at github.io

mfittipaldi90, 2023-10-16:
Thanks, that fixed it Just struggled a bit deleting old files. Had to figure out the permissions

closed 2023-10-16 by mfittipaldi90