kizniche/Mycodo #1342: Error: dialog not installed. Install it then try the Mycodo setup again.

qwoolsey, 2023-10-16:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

### Versions:

 - Mycodo Version: Latest from install link
 - Raspberry Pi Version: 4b
 - Raspbian OS Version: Lite 64-bit, just installed latest from site

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:
I installed mycodo using the github instructions, immediately after a fresh flash install of the latest 64-bit Lite Raspberry OS.
I receive an error during Mycodo install and have had no luck looking up the error dialog. 
Navigating to the ip address in a browser does not return the Mycodo UI and says no connection. 
However, the IP is pingable and I am connecting to headless pi using Putty.

Here is the install log of Mycodo.

curl -L https://kizniche.github.io/Mycodo/install | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current Dload  Upload   Total   Spent    Left  Speed
100  1293  100  1293    0     0   8590      0 --:--:-- --:--:-- --:--:--  8677
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
jq is already the newest version (1.6-2.1).
whiptail is already the newest version (0.52.23-1+b1).
python3 is already the newest version (3.11.2-1+b1).
wget is already the newest version (1.21.3-1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--2023-10-15 19:16:19--  https://api.github.com/repos/kizniche/Mycodo/tarball/v8.15.10
Resolving api.github.com (api.github.com)... 140.82.113.6
Connecting to api.github.com (api.github.com)|140.82.113.6|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/kizniche/Mycodo/legacy.tar.gz/refs/tags/v8.15.10 [following]
--2023-10-15 19:16:19--  https://codeload.github.com/kizniche/Mycodo/legacy.tar.gz/refs/tags/v8.15.10
Resolving codeload.github.com (codeload.github.com)... 140.82.113.9
Connecting to codeload.github.com (codeload.github.com)|140.82.113.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘mycodo-latest.tar.gz’

mycodo-latest.tar.gz                                            [   <=>                                                                                                                                   ]   6.14M  10.0MB/s    in 0.6s

2023-10-15 19:16:20 (10.0 MB/s) - ‘mycodo-latest.tar.gz’ saved [6439359]

FINISHED --2023-10-15 19:16:20--
Total wall clock time: 1.4s
Downloaded: 1 files, 6.1M in 0.6s (10.0 MB/s)
Checking Python version...
Python >= 3.6 found.

Error: dialog not installed. Install it then try the Mycodo setup again.

kizniche, 2023-10-16:
>whiptail is already the newest version (0.52.23-1+b1).

You are not attempting to install the latest Mycodo version if whiptail is in your code.

closed 2023-10-16 by kizniche

kizniche, 2023-10-16:
This was an oversight on my part with building the docs, which is how the install script is served with github.io. In any case, keep this issue closed and use #1341 for any future updates, since this is the same issue