kizniche/Mycodo #1344: Add ability to edit measurement database entries

kizniche, 2023-10-23:
Add admin page to edit measurement database entries, useful for correcting issues.

tomdotorg, 2023-11-13:
This would be tremendously useful to remove outliers, calibration values, etc.

kizniche, 2023-11-21:
I was referring to the database entries that defined measurement information, such as unique ID, unit, etc. of a measurement of an Input, Output, Function, not the measurements themselves. Influxdb has an interface of its own if you need to modify or delete specific measurements.

tomdotorg, 2023-11-25:
I see. I have had some occurrences of stray measurements that are way out of whack. I would love the ability to delete a single measurement without going through the learning curve and/or configuration complexity of learning Influx tooling for the RPi or enabling remote access to the database from the network. But I do understand that you do not want to reinvent the wheel here.