kizniche/Mycodo #1345: Dashboard camera Acquire Image (and save temporary file) permissions error

stminm, 2023-10-26:
### Describe the problem/bug

I'm using PiCamera. Capture Still, Timelaps and even dashboard widget with Acquire Image (and save new file) works fine.
Dashboard widget with Acquire Image (and save temporary file) creates and shows the image but it fails to update the new image at Refresh interval (30 sec).

Changing permissions to 777 for tmp file and the whole /tmp directory didn't fix the issue.
If I remove the /tmp/bde2d859-6afc-4758-9952-b2c5fdf991d3_tmp.jpg it will be recreated at next Refresh time but will not be updated.

### Versions:

 - Mycodo Version: 8.15.11
 - Raspberry Pi Version: Pi 3 Model B
 - Raspbian OS Version: 11 (bullseye)

### Reproducibility

Add Camera widget with Acquire Image (and save temporary file).
First image is fine and saved in /tmp/ as
-rw-r--r-- 1 mycodo mycodo 4836983 Oct 26 11:04 bde2d859-6afc-4758-9952-b2c5fdf991d3_tmp.jpg

The image is not refreshed although the image Timestamp in dashboard is changing.
In /var/log/daemon.log I see the output: 

Oct 26 11:23:10 ras-pi01 python[1712]: 2023-10-26 11:23:10,355 picamera
Oct 26 11:23:10 ras-pi01 python[1712]: Traceback (most recent call last):
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/home/pi/Mycodo/mycodo/devices/camera.py", line 138, in camera_record
Oct 26 11:23:10 ras-pi01 python[1712]:     camera.capture(path_file, use_video_port=False)
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/camera.py", line 1605, in capture
Oct 26 11:23:10 ras-pi01 python[1712]:     encoder.start(output)
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/encoders.py", line 1141, in start
Oct 26 11:23:10 ras-pi01 python[1712]:     super(PiCookedOneImageEncoder, self).start(output)
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/encoders.py", line 375, in start
Oct 26 11:23:10 ras-pi01 python[1712]:     self._open_output(output)
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/encoders.py", line 331, in _open_output
Oct 26 11:23:10 ras-pi01 python[1712]:     self.outputs[key] = mo.open_stream(output)
Oct 26 11:23:10 ras-pi01 python[1712]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/mmalobj.py", line 401, in open_stream
Oct 26 11:23:10 ras-pi01 python[1712]:     stream = io.open(stream, 'wb' if output else 'rb', buffering)
Oct 26 11:23:10 ras-pi01 python[1712]: PermissionError: [Errno 13] Permission denied: '/tmp/bde2d859-6afc-4758-9952-b2c5fdf991d3_tmp.jpg'

### Expected behavior

Image in /tmp should be replaced with a new one every 30 sec (default refresh time).

kizniche, 2023-10-26:
Thanks for the detailed bug report. I'll look into it.

kizniche, 2023-11-21:
Are you using the Camera Function or accessing the camera from the Camera page? Functions are where I'm moving all cameras to, and the Camera page will be phased out in a future release.

stminm, 2023-11-21:
I'm accessing camera from Camera page (More > Camera)

kizniche, 2023-12-19:
The master branch has a potential fix for this issue. If you would like to Upgrade to Master, you can test the code prior to release. You can find how to upgrade to master on the github wiki.

closed 2024-10-04 by kizniche