kizniche/Mycodo #1347: CAN Bus for sensors and relays

samuk, 2023-10-29:
**Is your feature request related to a problem? Please describe.**
I2C can be flaky, some of my systems may end up some distance from main compute module.

**Describe the solution you'd like**
Add software support for the CAN Bus components found in the open hardware [Acorn CM4 carrier](https://github.com/Twisted-Fields/acorn-robot-electronics/tree/main)

I _think_ this would enable CAN Bus communication to: 

- Open hardware [Olimex ESP32-EVB-Relay](https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware)
- Proprietary [I2C > CanBus modules](https://www.longan-labs.cc/1030017.html)
- Open hardware [I2C > CAN Bus daughter boards](https://www.electronics-lab.com/community/index.php?/topic/47903-i2c-to-can-bus-pcb-design/) (to be developed) for example a variant of the [Droneboard](https://cdn.shopify.com/s/files/1/0691/7072/5155/files/DroneBoard_1.0_DataSheet.pdf?v=1678677525) that supports CAN Bus rather than I2C over RJ45.

**Describe alternatives you've considered**
Using I2C over RJ45 as in the [Hiveboard/ Droneboard](https://hiveboard.ca/) project/ product

**Additional context**
Quite new to Canbus, so this may not be well formed or particularly coherent

FlooSz, 2023-10-31:
You can also use MQTT for good communication between Mycodo and e.g. Tasmota on a ESP,  if you only care about the RJ45 connector: It doesn't matter which cable you use for sensors or relays, as long as the cable is designed for the appropriate voltage, but for this side of the relay you could also connect an RJ45 socket to the I2C bus and use the same wires on the other side

Until there is a solution to your problem/upgrade, I would also find the topic interesting

FlooSz, 2023-10-31:
I can try to write you a code for a custom input, but we maybe must troubleshoote and and upgrade the code, I need for this all errors that will happen, and for starting I need to know which Sensors you want to use and which measurements will be required (Temp, Moisture,...and the units please)

samuk, 2023-10-31:
Thanks, I'm not particular about the cables. As you say they don't matter all that much with CAN Bus or I2C.

This is just at research phase at the moment, but good to know it seems possible. Initially I'd probably look to emulate the sensors from https://atlas-scientific.com/product/wi-fi-aquaponics-kit/

LucidEye, 2023-11-22:
@samuk 

If you need to increase the distance between your Raspberry Pi and your wired sensors there are a few options that are inexpensive and already available instead of having to develop hardware and software to adapt the CAN bus for Mycodo.

Adafruit has this active terminator which allows you to use a standard Ethernet cable as an "extension cord" (up to 100 feet) between the Pi and your i2c sensors.  It changes the signal for longer distances so you don't have to worry about things like line capacitance, pull-up resistors, or EMI.  There are several people here on the forum and on the Discord who are successfully using these active terminators to increase i2c distances.
https://www.adafruit.com/product/4756

Sparkfun also makes these i2c range extenders that use standard Ethernet cables as i2c extension cords (up to 100 feet).  They also use the STEMMA QT or QWIIC connectors which makes it easy to connect sensors.  You need to attach one Endpoint adapter to your Pi's i2c pins, and then use the Ethernet as the long cable run, and then have another endpoint adapter on the far end of the Ethernet cable, and that's where your sensors go...
https://www.sparkfun.com/products/16988

They also have these mid-point adapters for dropping sensors anywhere along the cable run.
https://www.sparkfun.com/products/18000

wbard93, 2023-11-23:
I've tried a few different I2C interfaces but have not been able to get it to be reliable. I'm currently working on a custom design using an ATMEGA328 on an interface board to go between the RPi I2C and an external RS485 bidirection bus.

I'm not familiar with CAN parts so it would be a while before I'd be able to look into that and experiment.

samuk, 2023-12-16:
The carrier board uses the standard MCP2515 which is in the Linux kernel

mickeprag, 2024-12-03:
For information, I am currently working on a MyCodo compatible board. It will use CAN for communication so I will spend some time to implement some sort of support for it in MyCodo.

victorhooi, 2025-03-14:
@mickeprag Just wondering how your board was coming along? Would definitely love to find out more! =)

mickeprag, 2025-03-14:
@victorhooi The progress is good. The design is finished and I have manufactured a bunch och boards and sent to different people interested in either helping our or just hack on their own. So far they seem to be working well... 🤞
If you are interested, you can find more information here:
https://docs.openhydroponics.com/hardware/rootmaster.html

But (there is always a but, isn't there?) the software integration into Mycodo is not yet complete. I have some issues I need help with before I can continue.

victorhooi, 2025-03-14:
Nice - I'll take a look through it =). (I'm not an electronics guru, but it looks pretty cool, and would love to try it/tinker with it later).

Also - filed a small PR to fix some documentation typos - https://gitlab.com/openhydroponics/sw/openhydroponics/-/merge_requests/3