kizniche/Mycodo #1349: Error 500: Internal Server Error

ControlFreakyKiwi, 2023-11-06:
Hi ,

Thanks for all your work on this project, I haven't revisited this until recently as I just purchased some raspberrypi shield relay to actually do something real..

Do I need to start from scratch and reinstall or can I salvage the situation?

Thanks in advance 

Carlito

'NoneType' object has no attribute 'split'

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.13.10
Database: 2c70a9e9e131
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Feb 25 2021 12:13:12
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 279, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

kizniche, 2023-11-21:
Duplicate of #1266. Please use the search before opening a new issue.

closed 2023-11-21 by kizniche

ControlFreakyKiwi, 2023-12-04:
Hi Kyle,
 
Ok sorry about that, I'm not that proficient with some of this stuff.
I ran that sed command which lead to a Error 502: Internal Server Error.
 
searched and searched but couldn't find a solution.
I deleted the Mycodo directory and started from scratch, same error.
I ended up entering the virtual environment and had to load all the below packages manually.
 
It seems to work now.
Vanilla Pi setup 
Can I help diagnose something?
 
 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
 
 
flask_login
flask_compress
flask_limiter
flask_session
flask_talisman
flask_sqlalchemy
flask_marshmallow
marshmallow-sqlalchemy
Pyro5
flask_wtf
distro
geocoder
influxdb
email_validator
flask_restx
flask_accept
filelock

kizniche, 2023-12-04:
The issue was resolved in January. You must not have updated your system since then. If you're installing Mycodo new, I would recommend installing the latest operating system. You appear to be running Buster.

ControlFreakyKiwi, 2023-12-04:
OK , thanks sorry for the confusion, keep up the great work

kizniche, 2023-12-05:
No problem. I'd rather hear about an issue if you're unsure if it's a duplicate than to potentially overlook it.