kizniche/Mycodo #1352: Permission denied when saving Conditional Controller Function

wbard93, 2023-11-23:
### Describe the problem/bug

*After installing a new Raspberry Pi OS and Mycodo, when saving settings in a Conditional Controller Function, a permission denied error is generated when creating a directoy*

### Versions:

 - Mycodo Version: *8.15.12*
 - Raspberry Pi Version: *3B+, 4B*
 - Raspbian OS Version: *RaspberryPi OS (Legacy) Lite (32 and 64 bit)*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *Use Raspberry Pi Imager v1.7.4 on Mac to copy Raspberry Lite 2023-10-10 to 16GB Micro SD card*
2. *Boot Raspberry Pi, connect to it with the Terminal app and enter 'curl -L https://kizniche.github.io/Mycodo/install | bash' to install Mycodo*
3. *Create the Mycodo admin account and log in*
4. *Select Function menu, create Conditional Controller (installing dependencies), and open created Conditional Controller Function*
5. *When clicking the Save button on the opened Conditional Controller Function, the error appears*

*Nothing seen in Mycodo logs. Not sure what other logs to check.*

*Not a unix expert but do not see any folders or files corresponding to error message in a working or non-working system.*

### Expected behavior

*No error should occur*
*First, after clicking the Save button, the command sent message appears*
*The after about 30 seconds, the warning, info and success messages appear*

### Screenshots

*Expected condition after clicking the Save button
*First, after clicking the Save button, the command sent message appears*
*[image]*
*The after about 30 seconds, the warning, info and success messages appear*
*[image]*

*Failed condition after clicking the Save button
*[image]*

### Additional context

*This works on Mycodo 8.15.12 that has been upgraded. Problem only occurs on a new install*
*As a workaround, looking to duplicate a working Micro SD card and using that.*

kizniche, 2023-11-23:
This is the same issue fixed in 72c1dde7548754dd1aaad56e551ee2fc145d0a1f for Python Code Inputs. I'll apply the same fix for Conditional Controller Functions and make a bugfix release.

kizniche, 2023-11-23:
To be clear, this doesn't affect the functionality of a Conditional Controller, it merely prevents pylint from running.

closed 2024-10-04 by kizniche