kizniche/Mycodo #1353: request to add Adafruit NeoKey 1x4 input board

Nithalik, 2023-12-10:
Hi Kyle,

Found a device from adafruit that would be able to add a lot of possible functionality for people not super skilled in programming if you can add this input device.  It is basically a i2c controlled mini keyboard that has 4 keyboard keys and 4 neopixels. basically a easy way to add 4 i2c controlled momentary switches to a system.  It also has 8 possible i2c addresses so there is a lot of possibility here

https://www.adafruit.com/product/4980

adafruit neokey library: https://docs.circuitpython.org/projects/neokey/en/latest/

i have ordered one and been playing with it, have gotten their test code working, but getting it to work with mycodo is not working for me

kizniche, 2023-12-10:
How were you envisioning it integrating?

Nithalik, 2023-12-10:
im wanting to use the switches to start functions, and change what is displayed on a different i2c screen. neopixels are just a added bonus, they could be used as status led's or to show that the button was pressed and recognized

like be able to have a button to trigger some of the input actions on this page https://kizniche.github.io/Mycodo/Inputs/#input-actions

the more useful ones would be to trigger the supported actions which is where i would use it the most

https://kizniche.github.io/Mycodo/Supported-Actions/

button to capture a pic from all the cameras / execute command: shell / display backlight  /  pause  would be the big ones i can think of right away.

im not currently able to access my mycodo system so not able to really go through all the different sections to see where it is useful, but i envision using it so i dont have to go into the webui when i do maintenance, i can do it by buttons.

-pause/enable everything
-run waterchange function
-pH balance / add nutrients
-turn off main lights (or dim)

kizniche, 2023-12-10:
I could throw together a simple function that you could execute one or more functions when a button is pressed.

Nithalik, 2023-12-10:
that would work for almost all of my use cases, and i think i can add a lot with that.  the only other thing is if there would be a way to address the neoled's too.

example for 1 button - when button is pressed, runs action, then flashes the neoled while that is active. and push the button again to undo if its not one based on a timer.

kizniche, 2023-12-10:
All three sound doable. I'll let you know when I have a module for you to test.

Nithalik, 2023-12-10:
> All three sound doable. I'll let you know when I have a module for you to test.

cool,  thanks

kizniche, 2023-12-10:
Attached below is a module to test. Rename from txt to py and import as a Function. You will also need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to be able to use Actions for Custom Functions and see the Action ID needed to use the new Function. The instructions after creating the Function should be sufficient to use it. I won't detail how to use it to see if the instructions can stand on their own. Let me know if you run into any issues or can't get it working. Remember to enable the Function debug logging when testing to see all debug messages in the log.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13629436/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-10:
Thank you for that, got it installed.

when i push a button it just lights up the LED underneath but nothing else. below is the error log.

just have all buttons mapped to deactivate a function for testing.

```python

``` AttributeError: 'list' object has no attribute 'unique_id'
2023-12-10 15:31:46,916 - ERROR - mycodo.function.function_adafruit_neokey_01_8df033ec - Executing Action with ID dd3d8ad0
Traceback (most recent call last):
  File "/home/nithalik/Mycodo/mycodo/functions/custom_functions/function_adafruit_neokey_01.py", line 236, in check_key_press
    action.unique_id,
AttributeError: 'list' object has no attribute 'unique_id'
2023-12-10 15:31:46,928 - INFO - mycodo.function.function_adafruit_neokey_01_8df033ec - Button 4
2023-12-10 15:31:47,461 - ERROR - mycodo.function.function_adafruit_neokey_01_8df033ec - Executing Action with ID 8ceb33e8
Traceback (most recent call last):
  File "/home/nithalik/Mycodo/mycodo/functions/custom_functions/function_adafruit_neokey_01.py", line 236, in check_key_press
    action.unique_id,
AttributeError: 'list' object has no attribute 'unique_id'
2023-12-10 15:31:47,473 - ERROR - mycodo.function.function_adafruit_neokey_01_8df033ec - Executing Action with ID d05c3d60
Traceback (most recent call last):
  File "/home/nithalik/Mycodo/mycodo/functions/custom_functions/function_adafruit_neokey_01.py", line 236, in check_key_press
    action.unique_id,
AttributeError: 'list' object has no attribute 'unique_id'
2023-12-10 15:31:47,486 - ERROR - mycodo.function.function_adafruit_neokey_01_8df033ec - Executing Action with ID dd3d8ad0
Traceback (most recent call last):
  File "/home/nithalik/Mycodo/mycodo/functions/custom_functions/function_adafruit_neokey_01.py", line 236, in check_key_press
    action.unique_id,
AttributeError: 'list' object has no attribute 'unique_id'

kizniche, 2023-12-10:
Are you sure you've upgraded to master?

kizniche, 2023-12-10:
Your screenshot does not look correct. Your Function is missing all options to set for each key.

kizniche, 2023-12-10:
Here is an updated module. Delete all of this Function type, then delete the Function module, then import this new Function module.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13629601/function_adafruit_neokey_01.txt)

kizniche, 2023-12-10:
Wait, you didn't add the correct Function. That's one of the built-in functions.

Nithalik, 2023-12-10:
the screenshot was just the execute actions, to show the id's

in changing the settings in it, now the method page wont load giving 500 error. i must have done something stupid. reinstalling mycodo and will test out the new file

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/routes_function.py", line 514, in page_function
    return render_template('pages/function.html',
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/templating.py", line 151, in render_template
    return _render(app, template, context)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/templating.py", line 132, in _render
    rv = template.render(context)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Functions/", dict_translation['function']['title']] %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 383, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 135, in block 'body'
    {% include 'pages/function_options/custom_function_entry.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/custom_function_entry.html", line 51, in top-level template code
    {% include 'pages/function_options/custom_function_options.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/custom_function_options.html", line 292, in top-level template code
    {% include 'pages/actions.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/actions.html", line 40, in top-level template code
    {% include 'pages/form_options/Custom_Options.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/form_options/Custom_Options.html", line 285, in top-level template code
    <option value="{{each_dev.unique_id}}"{% if (channel_options and each_dev.unique_id == custom_options_values[unique_id][channel][each_option['id']]) or (custom_options_values and each_dev.unique_id == custom_options_values[unique_id][each_option['id']]) %} selected{% endif %}>[{{name}} {{'%02d' % each_dev.id}}] {{each_dev.name}}</option>
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 466, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: dict object has no element Undefined

kizniche, 2023-12-10:
The IDs should show on all Functions. You should be only using the FUnction I created.

kizniche, 2023-12-10:
What did you do between it working and not working?

Nithalik, 2023-12-10:
all i did was change the ID's in the new module to random numbers, thinking that using the id's from the execute actions function was causing the issue

Nithalik, 2023-12-10:
i was just showing with that screenshot that i was didnt make a error copying the action id's into the boxes in the new module

kizniche, 2023-12-10:
Your error is resuting from a Select option issue, but there are no select options in the module I gave you, so the module isn't causing that error.

kizniche, 2023-12-10:
It would be most beneficial if you can give steps to reproduce any errors you get. If you found an error that crashes the frontend, that is a major issue.

Nithalik, 2023-12-10:
> It would be most beneficial if you can give steps to reproduce any errors you get. If you found an error that crashes the frontend, that is a major issue.

i am working on trying to reproduce it now, as i could not load the method page, not much i could do other then reinstall. which i am doing now.  if i can reproduce it, i will provide steps

all i did, was change the ids in the new module, not sure how that changed other things.

Nithalik, 2023-12-11:
pretty sure i see what causes it, fresh install, installed the new module, and clicked on add action in the module itself, after adding a module to deactivate a controller in the action itself it gave the 500 error

/Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/routes_function.py", line 514, in page_function
    return render_template('pages/function.html',
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/templating.py", line 151, in render_template
    return _render(app, template, context)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/templating.py", line 132, in _render
    rv = template.render(context)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 3, in top-level template code
    {% set help_page = ["https://kizniche.github.io/Mycodo/Functions/", dict_translation['function']['title']] %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 383, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function.html", line 135, in block 'body'
    {% include 'pages/function_options/custom_function_entry.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/custom_function_entry.html", line 51, in top-level template code
    {% include 'pages/function_options/custom_function_options.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/function_options/custom_function_options.html", line 292, in top-level template code
    {% include 'pages/actions.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/actions.html", line 40, in top-level template code
    {% include 'pages/form_options/Custom_Options.html' %}
  File "/home/nithalik/Mycodo/mycodo/mycodo_flask/templates/pages/form_options/Custom_Options.html", line 285, in top-level template code
    <option value="{{each_dev.unique_id}}"{% if (channel_options and each_dev.unique_id == custom_options_values[unique_id][channel][each_option['id']]) or (custom_options_values and each_dev.unique_id == custom_options_values[unique_id][each_option['id']]) %} selected{% endif %}>[{{name}} {{'%02d' % each_dev.id}}] {{each_dev.name}}</option>
  File "/var/mycodo-root/env/lib/python3.9/site-packages/jinja2/environment.py", line 466, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: dict object has no element Undefined

Nithalik, 2023-12-11:
Got it somewhat working.

new mycodo system. 
[image]

all the activate and deactivate does is turn on or off the ph and ec regulation.

the buttons work to turn it on and off. the only issue is, when using the deactivate action, the led never turns off. when using activate the led turns on for 1 second

[image]

kizniche, 2023-12-11:
I was able to replicate the error. Looking into a fix.

Nithalik, 2023-12-11:
all working now, and the code change worked.

there are cases where having it be enabled for total duration would be helpful, would it be difficult to be able to select per key if it is pixels[0] to pixels[key]

kizniche, 2023-12-11:
What do you mean by enabled for total duration? The LED stays on for as long as it takes the Actions to complete.

I also just majorly refactored the Function Module to utilize another method for organizing configuration data so it's easier to work with in the code. I also made the key press spawn a thread to complete the actions, so you can now have multiple key action sets running at the same time and you don't have to wait for the first to finish before starting the next.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630092/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> Your screenshot does not look correct. Your Function is missing all options to set for each key.

i did upgrade to master. the screentshot was just of the execute actions of the functions i was going to use, to show the id's

> What do you mean by enabled for total duration? The LED stays on for as long as it takes the Actions to complete.
> 
> I also just majorly refactored the Function Module to utilize another method for organizing configuration data so it's easier to work with in the code. And added a delay to keep the LED on after all actions have completed. I also made the key press spawn a thread to complete the actions, so you can now have multiple key action sets running at the same time and you don't have to wait for the first to finish before starting the next.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630092/function_adafruit_neokey_01.txt)

i just misunderstood the code change. what you did is perfect, the new changes make it very clear what is happening

Nithalik, 2023-12-11:
> I'm working on allowing the color of each key to be set

if your working on that side, is it possible to have a start colour, then it change colour on its last action? like a colour change to signify when it is done running the actions. just a thought

kizniche, 2023-12-11:
Here is an update that adds LED colors for resting, running, and last action running. Let me know if it works.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630286/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> Here is an update that adds LED colors for resting, running, and last action running. Let me know if it works.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630297/function_adafruit_neokey_01.txt)

very nice just tested and it all works.

only thing i would add, is since there is a resting led state now, maybe add a brightness option to resting so you can have the light off on rest

kizniche, 2023-12-11:
I don't think brightness can be set. Here is a version that turns the LED off when set to 0.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630470/function_adafruit_neokey_01.txt)

kizniche, 2023-12-11:
This version may or may not work to set the brightness.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630510/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
Just running into a bit of a issue,

It does not seem like the buttons are actually doing anything now. the logs show the button press, but not that it is triggering an action, have saved it and tested a few times

2023-12-10 19:06:00,746 - INFO - mycodo.controllers.controller_function_7731aef4 - Activated in 710.7 ms
2023-12-10 19:06:01,331 - INFO - mycodo.function.function_adafruit_neokey_01_7731aef4 - Key 1
2023-12-10 19:06:02,441 - INFO - mycodo.function.function_adafruit_neokey_01_7731aef4 - Key 2
2023-12-10 19:06:03,276 - INFO - mycodo.function.function_adafruit_neokey_01_7731aef4 - Key 3
2023-12-10 19:06:05,563 - INFO - mycodo.function.function_adafruit_neokey_01_7731aef4 - Key 4
2023-12-10 19:06:09,512 - INFO - mycodo.controllers.controller_function_7731aef4 - Deactivated in 125.4 ms
2023-12-10 19:07:07,837 - DEBUG - mycodo.controllers.controller_function_7731aef4 - loop() not found
2023-12-10 19:07:07,839 - DEBUG - mycodo.controllers.controller_function_7731aef4 - listener() found
2023-12-10 19:07:07,839 - DEBUG - mycodo.controllers.controller_function_7731aef4 - Starting listener() thread
2023-12-10 19:07:07,840 - INFO - mycodo.controllers.controller_function_7731aef4 - Activated in 717.4 ms
2023-12-10 19:07:27,870 - INFO - mycodo.function.function_adafruit_neokey_01_7731aef4 - Key 1

kizniche, 2023-12-11:
Here is a version with more debugging lines to find out what's wrong. Press a key that has an action associated with it and watch the log for debug lines.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630579/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> This version may or may not work to set the brightness.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630528/function_adafruit_neokey_01.txt)

this results in the below error, brightness is not an issue, it was more having the led off when not needed

SError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 181, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/busio.py", line 197, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=True)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 67, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 170, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nithalik/Mycodo/mycodo/functions/custom_functions/function_adafruit_neokey_01.py", line 274, in initialize
    self.neokey = NeoKey1x4(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_neokey/neokey1x4.py", line 57, in __init__
    super().__init__(i2c_bus, addr)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_seesaw/seesaw.py", line 147, in __init__
    self.i2c_device = I2CDevice(i2c_bus, addr)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 62, in __init__
    self.__probe_for_device()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 184, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x30
2023-12-10 19:11:39,586 - INFO - mycodo.controllers.controller_function_d52ad914 - Activated in 194.5 ms

kizniche, 2023-12-11:
I don't think this is a result of the brightness code I put in: "ValueError: No I2C device at address: 0x30"

kizniche, 2023-12-11:
Are you sure you saved the action options?

Nithalik, 2023-12-11:
> Are you sure you saved the action options?

yes, i ensured i saved, and also restated the rpi to make sure

kizniche, 2023-12-11:
Try this with the debug lines moved to try to understand more.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630625/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
the logs now are not even showing the button press on the most recent file. only shows activating and deactivating the controller
```python
2023-12-10 19:27:27,355 - INFO - mycodo.controllers.controller_function_2f7d090c - Deactivated in 234.4 ms
2023-12-10 19:27:29,367 - DEBUG - mycodo.function.function_adafruit_neokey_01_2f7d090c - Parsed Action IDs: OrderedDict([(0, ['e793b14e']), (1, ['5e9e8958']), (2, []), (3, [])])
2023-12-10 19:27:29,906 - DEBUG - mycodo.controllers.controller_function_2f7d090c - loop() not found
2023-12-10 19:27:29,907 - DEBUG - mycodo.controllers.controller_function_2f7d090c - listener() found
2023-12-10 19:27:29,907 - DEBUG - mycodo.controllers.controller_function_2f7d090c - Starting listener() thread
``` 

was the most recent before the upgrade

kizniche, 2023-12-11:
I didn't change anything except move a log line. So there's something not working about the device if it's not detectng button presses.

kizniche, 2023-12-11:
Are you pressing the button for long enough?

Nithalik, 2023-12-11:
yes the panel recognizes the button press, the led even turns on to what it is supposed to be for the running led colour, it just stays at that colour and nothing changes in the log

Nithalik, 2023-12-11:
i went back to the version before you added the different led colour options and everything works right away

> What do you mean by enabled for total duration? The LED stays on for as long as it takes the Actions to complete.
>
> I also just majorly refactored the Function Module to utilize another method for [...]

I went back to this version and everything is working again

kizniche, 2023-12-11:
Here's one with many more debig lines. Are you deleting the Function then deleting the Function Module, then importing the new module, then adding the Function?

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630730/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> Here's one with many more debig lines. Are you deleting the Function then deleting the Function Module, then importing the new module, then adding the Function?
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630730/function_adafruit_neokey_01.txt)

yes i am following that process, delete the function, then the function module, then importing and adding new function

Nithalik, 2023-12-11:
here is the error log from the most 

> Here's one with many more debug lines. Are you deleting the Function then deleting the Function Module, then importing the new module, then adding the Function?
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630730/function_adafruit_neokey_01.txt)

here is the error log from this one, buttons still do not seem to do anything, have to go to the old file again

```python
2023-12-10 19:47:07,584 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Parsed Action IDs: OrderedDict([(0, ['8e4a9986']), (1, ['34505ec9']), (2, []), (3, [])])
2023-12-10 19:47:07,584 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Brightness: 0.2
2023-12-10 19:47:07,585 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Colors: OrderedDict([(0, 0), (1, 0), (2, 0), (3, 0)]), OrderedDict([(0, 85), (1, 85), (2, 85), (3, 85)]), OrderedDict([(0, 255), (1, 255), (2, 255), (3, 255)])
2023-12-10 19:47:08,122 - DEBUG - mycodo.controllers.controller_function_eed44e0c - loop() not found
2023-12-10 19:47:08,124 - DEBUG - mycodo.controllers.controller_function_eed44e0c - listener() found
2023-12-10 19:47:08,125 - DEBUG - mycodo.controllers.controller_function_eed44e0c - Starting listener() thread
2023-12-10 19:47:08,126 - INFO - mycodo.controllers.controller_function_eed44e0c - Activated in 910.0 ms
2023-12-10 19:47:13,142 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Key 2
2023-12-10 19:47:13,663 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Executing Actions for Key 1
2023-12-10 19:47:18,630 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Key 1
2023-12-10 19:47:19,152 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Executing Actions for Key 0
2023-12-10 19:47:25,412 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Key 2
2023-12-10 19:47:26,451 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Executing Actions for Key 1
2023-12-10 19:47:27,404 - DEBUG - mycodo.function.function_adafruit_neokey_01_eed44e0c - Key 1
```

kizniche, 2023-12-11:
Now we're getting somewhere. Try this version. I suspect we'll see something strange in this one.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630742/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> Now we're getting somewhere. Try this version. I suspect we'll see something strange in this one.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630742/function_adafruit_neokey_01.txt)

Here are the logs from that one

```python
2023-12-10 19:54:10,793 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Parsed Action IDs: OrderedDict([(0, ['34505ec9']), (1, ['8e4a9986']), (2, []), (3, [])])
2023-12-10 19:54:10,794 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Brightness: 0.2
2023-12-10 19:54:10,794 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Colors: OrderedDict([(0, 0), (1, 0), (2, 0), (3, 0)]), OrderedDict([(0, 85), (1, 85), (2, 85), (3, 85)]), OrderedDict([(0, 255), (1, 255), (2, 255), (3, 255)])
2023-12-10 19:54:11,332 - DEBUG - mycodo.controllers.controller_function_b6dc24df - loop() not found
2023-12-10 19:54:11,332 - DEBUG - mycodo.controllers.controller_function_b6dc24df - listener() found
2023-12-10 19:54:11,333 - DEBUG - mycodo.controllers.controller_function_b6dc24df - Starting listener() thread
2023-12-10 19:54:11,334 - INFO - mycodo.controllers.controller_function_b6dc24df - Activated in 712.1 ms
2023-12-10 19:54:12,799 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Key 2
2023-12-10 19:54:13,327 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Executing Actions for Key 2
2023-12-10 19:54:18,227 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Key 1
2023-12-10 19:54:18,749 - DEBUG - mycodo.function.function_adafruit_neokey_01_b6dc24df - Executing Actions for Key 1
``` 

here are the logs from the last version that was working
```python
2023-12-10 19:51:34,573 - INFO - mycodo.controllers.controller_function_11107a50 - Deactivated in 230.7 ms
2023-12-10 19:51:41,250 - DEBUG - mycodo.controllers.controller_function_11107a50 - loop() not found
2023-12-10 19:51:41,251 - DEBUG - mycodo.controllers.controller_function_11107a50 - listener() found
2023-12-10 19:51:41,252 - DEBUG - mycodo.controllers.controller_function_11107a50 - Starting listener() thread
2023-12-10 19:51:41,262 - INFO - mycodo.controllers.controller_function_11107a50 - Activated in 731.0 ms
2023-12-10 19:51:51,647 - INFO - mycodo.function.function_adafruit_neokey_01_11107a50 - Key 1
2023-12-10 19:51:52,401 - DEBUG - mycodo.function.function_adafruit_neokey_01_11107a50 - Return message:  Deactivate Controller 4671a538-ed2d-4c17-888e-00eaab4414c0 (Seedling Regulate pH and Electrical Conductivity).
2023-12-10 19:51:52,699 - INFO - mycodo.controllers.controller_function_4671a538 - Deactivated in 210.6 ms
2023-12-10 19:52:00,551 - INFO - mycodo.function.function_adafruit_neokey_01_11107a50 - Key 2
2023-12-10 19:52:01,247 - DEBUG - mycodo.function.function_adafruit_neokey_01_11107a50 - Return message:  Activate Controller 4671a538-ed2d-4c17-888e-00eaab4414c0 (Seedling Regulate pH and Electrical Conductivity).
2023-12-10 19:52:01,691 - INFO - mycodo.controllers.controller_function_4671a538 - Activated in 356.7 ms
2023-12-10 19:52:07,173 - INFO - mycodo.function.function_adafruit_neokey_01_11107a50 - Key 1
2023-12-10 19:52:07,866 - DEBUG - mycodo.function.function_adafruit_neokey_01_11107a50 - Return message:  Deactivate Controller 4671a538-ed2d-4c17-888e-00eaab4414c0 (Seedling Regulate pH and Electrical Conductivity).
2023-12-10 19:52:08,205 - INFO - mycodo.controllers.controller_function_4671a538 - Deactivated in 245.7 ms
```

kizniche, 2023-12-11:
I understand. There's nothing I can tell between the versions that would affect anything. THe only thing I cna do is give you versions to try, since i don't have the device.

kizniche, 2023-12-11:
This version I threw in a slew of degub lines.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630774/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> This version I threw in a slew of debug lines.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630774/function_adafruit_neokey_01.txt)

apologies, didnt know if you saw my message on the old version working. here are the new logs.

```python
2023-12-10 20:01:40,956 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Parsed Action IDs: OrderedDict([(0, ['34505ec9']), (1, ['8e4a9986']), (2, []), (3, [])])
2023-12-10 20:01:40,956 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Brightness: 0.2
2023-12-10 20:01:40,957 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Colors: OrderedDict([(0, 0), (1, 0), (2, 0), (3, 0)]), OrderedDict([(0, 85), (1, 85), (2, 85), (3, 85)]), OrderedDict([(0, 255), (1, 255), (2, 255), (3, 255)])
2023-12-10 20:01:41,494 - DEBUG - mycodo.controllers.controller_function_d40c3f98 - loop() not found
2023-12-10 20:01:41,495 - DEBUG - mycodo.controllers.controller_function_d40c3f98 - listener() found
2023-12-10 20:01:41,495 - DEBUG - mycodo.controllers.controller_function_d40c3f98 - Starting listener() thread
2023-12-10 20:01:41,496 - INFO - mycodo.controllers.controller_function_d40c3f98 - Activated in 715.6 ms
2023-12-10 20:01:53,894 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Key 2 Pressed
2023-12-10 20:01:54,423 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Executing Actions for Key 2
2023-12-10 20:01:54,423 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - TEST00
2023-12-10 20:02:03,045 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Key 1 Pressed
2023-12-10 20:02:03,566 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - Executing Actions for Key 1
2023-12-10 20:02:03,567 - DEBUG - mycodo.function.function_adafruit_neokey_01_d40c3f98 - TEST00
```

kizniche, 2023-12-11:
oh wow, it was the easiest fix. Simple error. Should work now.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630806/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
> oh wow, it was the easiest fix. Simple error. Should work now.
> 
> [function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13630806/function_adafruit_neokey_01.txt)

it is working now. everything seems to be working properly.

what was the issue?

kizniche, 2023-12-11:
"len(list)" instead of the actual list. "list" is a type, not a variable.

kizniche, 2023-12-11:
The last thing I can think of to add is to write a value of 1 to the measurement database for the channel of the key pressed, so it's recorded. We've been at it for a while, so I don't expect you to test thir right away if you're done for now. Just let me know if it works if you get a chance to test it.

[function_adafruit_neokey_01.txt](https://github.com/kizniche/Mycodo/files/13631004/function_adafruit_neokey_01.txt)

Nithalik, 2023-12-11:
we have success, always willing to test when i am the only one who currently benefits, appreciate the help here

[image]

kizniche, 2023-12-11:
I can't seem to get the name to appear on the chart without some refactoring, so I'm going to call it a wrap for now. Thanks for all the testing, I'll add this in my next commit to github.

kizniche, 2023-12-11:
The last thing I changed (and will commit soon) was figuring out how to have the key number show on measurements (such as on charts, live page, etc). I'm also going to implement an action to set the color of a neopixel at position x of a function and set this new function as the recipient module for that action (as well as the other neopixel functions). That way you can have the lights change in response to things elsewhere in the system, such as turning a key red to notify of a system issue.

If anyone else has ideas for improving the functionality, let me know and we can discuss the feasibility.

Nithalik, 2023-12-11:
the addon part is what i was initially envisioning, same button to turn on grow lights, then if on, the same button turns them off. thought it would be to complicated, but it does cut the need for a some of the keys down.

im thinking on how to expand this to use like 4-5 of the keypads.

kizniche, 2023-12-11:
Thankfully, I refactored the system a while ago to make development much quicker by allowing all code for a function to be contained in a single file, so developing these functions is much more enjoyable than previously (editing multiple files). This makes it rather easy to implement these features, as long as you understand how the system works and options available to build a module. My only limitation is not owning the hardware, which significantly slows testing. I just bought the board, so we soon won't have to have to share a dozen module files to make development progress.

[10 more replies]

closed 2024-10-04 by kizniche