kizniche/Mycodo #1354: Request to add DFRobot i2c DAC output module (0-10v)

Nithalik, 2023-12-10:
Found a awesome DAC from DFrobot, it is the first i2C device that i have found that outputs a 0-10v signal. also has 8 possible i2c addresses. i am a failure at programming so hoping this is universal enough for other people to find useful too.  

https://www.dfrobot.com/product-2613.html

tons of lighting systems use 0-10v dimming, and this is the easiest way to add it to mycodo without making your own circuit, which is how i am currently doing it, this is just more user friendly

https://www.prolighting.com/blog/2020/03/25/understanding-0-10v-dimming-with-todays-led-commercial-fixtures/

wiki: https://wiki.dfrobot.com/SKU_DFR0971_2_Channel_I2C_0_10V_DAC_Module

probably somewhat similar to https://kizniche.github.io/Mycodo/Supported-Outputs/#digital-to-analog-converter-mcp4728

i envision the output to just have the option to select i2c address and set output voltage, bit simpler then the mcp4728

they also have 6 version of this board with various different outputs

kizniche, 2023-12-10:
Do you own one that you can test with?

Nithalik, 2023-12-10:
i am ordering one today to test with.

Nithalik, 2023-12-10:
don't think it would matter from your side, but i can always order one of their different versions for testing if you are interested in it.

i just dont really understand the use of the (4-20mA) ones, i think there are really just commercial applications, but there are also the PWM ones

kizniche, 2023-12-10:
No need to buy others, unless you have a use for it.

Nithalik, 2023-12-10:
ok, ordered, should be here within 2 weeks

Nithalik, 2023-12-12:
asking here to see if this is even something feasible if this output module gets added to the system

if this output module is added, would it be hard to also add Trigger Configuration: Trigger: Output based on

[image]

if this output gets added, would it be hard to add a similar trigger output based on this output device? does not seem like you can trigger things by i2c devices at this point

kizniche, 2023-12-19:
It's feasible and wouldn't be too hard to develop. It would act like any other Value Output that allows you to set a value (between 0 and 10) and would work with the current Output Value Action.

Nithalik, 2023-12-22:
the parts have arrived.  have 2 of 2-Channel I2C DAC Module (0-10V) and 1 1-Channel 15-bit I2C to 0-5V/10V DAC Module

silverhawk1983, 2024-02-14:
that could open some options for both Lights and Fan Control  hope we can test some soon with that chips

silverhawk1983, 2024-02-27:
> It's feasible and wouldn't be too hard to develop. It would act like any other Value Output that allows you to set a value (between 0 and 10) and would work with the current Output Value Action.

0-1000 would be more suited for that kind of output in relation to 0.00v to 10.00v  so we could send  for  6.34v  - 634 as value 
internally  would be higher due to  12/15bit conversion internally 

realworld   0.00-10.00v
Mycodo 0-1000
module 12bit - 0-4.095 (0x000-0xFFF) / 15bit 0 - 32767 (0x0000-0x7FFF)

silverhawk1983, 2024-03-06:
i got 2 2ch 15bit 0-10v modules today in the mail did we have by any chance some code to test with them

Nithalik, 2024-03-06:
no code from me, am in the middle of a move

kizniche, 2024-03-06:
Yes, I have a module that you can test. I'll push the code later when I have access to my development system.

kizniche, 2024-03-06:
I just committed an untested module. Give it a test and let me know if there are any issues.

silverhawk1983, 2024-03-07:
will try it hope it will work with the 15bit to

silverhawk1983, 2024-03-07:
So i tryed use the new module with the 15bit thats not working but throws no error 

so  then i trid modifiy the module code for the GP8413 the 15bit one 
[value_p8413_dac_0_10_vdc.txt](https://github.com/kizniche/Mycodo/files/14527126/value_p8413_dac_0_10_vdc.txt)
could not find a lib for the 8413  so tryd with the 8403 but it trows 
mycodo.outputs.output_gp8413_dac_0_10_vdc_a3f3677f - Cannot manipulate Output a3f3677f-482a-4ada-9f5d-934557eb22d1: output channel doesn't exist: 0

silverhawk1983, 2024-03-07:
could that be used 
[https://github.com/DFRobot/DFRobot_GP8XXX/blob/master/python/raspberryPi/README.md](https://github.com/DFRobot/DFRobot_GP8XXX/blob/master/python/raspberryPi/README.md)

kizniche, 2024-03-07:
Have you tested the hardware the module was designed for? Let's stick to the original issue before bringing in other hardware into the discussion.

silverhawk1983, 2024-03-07:
sure your right  i have only the 15bit Version   but the Control is the same for all of those

only dif is the internal command for the value incase of 12 or 15 bit  showed that last week in a post 
the pipi lib is only to card coded only for the 8403 the DFrobot lib   has them all in them 

Produkt Overview shop
[https://www.dfrobot.com/product-2756.html](https://www.dfrobot.com/product-2756.html)

Produkt Overview wiki
[(https://wiki.dfrobot.com/SKU_DFR1073_2_Channel_15bit_I2C_to_0-10V_DAC)](https://wiki.dfrobot.com/SKU_DFR1073_2_Channel_15bit_I2C_to_0-10V_DAC)

DFRobot Github Lib
[https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi](https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi)

Joe2824  the lib you used  linked to Pip-pi
[https://github.com/Joe2824/DFRobot_GP8403/tree/main](https://github.com/Joe2824/DFRobot_GP8403/tree/main)

my Old reply to one of your replys
> > It's feasible and wouldn't be too hard to develop. It would act like any other Value Output that allows you to set a value (between 0 and 10) and would work with the current Output Value Action.
> 
> 0-1000 would be more suited for that kind of output in relation to 0.00v to 10.00v so we could send for 6.34v - 634 as value internally would be higher due to 12/15bit conversion internally
> 
> realworld 0.00-10.00v Mycodo 0-1000(equal to 0 to 1000mV) module 12bit - 0-4.095 (0x000-0xFFF) / 15bit 0-32767 (0x0000-0x7FFF)

silverhawk1983, 2024-03-07:
sure your right  i have only the 15bit Version   but the Control is the same for all of those

only dif is the internal command for the value in Case of 12 or 15 bit etc  showed that last week in a post 
the pip-pi lib is only to hard coded only for the 8403 the DFrobot lib instead has them all in it. 

Produkt Overview shop
[https://www.dfrobot.com/product-2756.html](https://www.dfrobot.com/product-2756.html)

Produkt Overview wiki
[(https://wiki.dfrobot.com/SKU_DFR1073_2_Channel_15bit_I2C_to_0-10V_DAC)](https://wiki.dfrobot.com/SKU_DFR1073_2_Channel_15bit_I2C_to_0-10V_DAC)

Link to Datasheets 
8403
[https://dfimg.dfrobot.com/nobody/wiki/0e324f3dbbd888ea78b2c0d23340998c.pdf](https://dfimg.dfrobot.com/nobody/wiki/0e324f3dbbd888ea78b2c0d23340998c.pdf)

8413
[https://dfimg.dfrobot.com/60c1e008bddfc41c3293de80/wiki/56397a22232d7e06df627ec8a688539a.pdf](https://dfimg.dfrobot.com/60c1e008bddfc41c3293de80/wiki/56397a22232d7e06df627ec8a688539a.pdf)

Joe2824  the lib you used  linked to Pip-pi
[https://github.com/Joe2824/DFRobot_GP8403/tree/main](https://github.com/Joe2824/DFRobot_GP8403/tree/main)

DFRobot Github Lib
[https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi](https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi)

my Old reply to one of your replys
> 0-1000 would be more suited for that kind of output in relation to 0.00v to 10.00v so we could send for 6.34v - 634 as value internally would be higher due to 12/15bit conversion internally [...]

kizniche, 2024-03-07:
And does the code you linked to work?

kizniche, 2024-03-07:
I don't think yur library is a good solurion. For one, it uses RPi.GPIO, which A) is not supported in Pi 5s, and B) is a terrible way to conduct I2C communication due to it changing the I2C pins to inputs and other issues that conflict with other libraries, such as smbus2.

silverhawk1983, 2024-03-08:
> I don't think your library is a good solution. For one, it uses RPi.GPIO, which A) is not supported in Pi 5s, and B) is a terrible way to conduct I2C communication due to it changing the I2C pins to inputs and other issues that conflict with other libraries, such as smbus2.

that´s true i linked that only for comparison

silverhawk1983, 2024-03-08:
> And does the code you linked to work?

no!

> So i try use the new module with the 15bit thats not working but throws no error
> 
> so then i did a modifiy of the module code for the GP8413 the 15bit one [value_p8413_dac_0_10_vdc.txt](https://github.com/kizniche/Mycodo/files/14527126/value_p8413_dac_0_10_vdc.txt) could not find a lib for the 8413 gone for the 8403 but it trows mycodo.outputs.output_gp8413_dac_0_10_vdc_a3f3677f - Cannot manipulate Output a3f3677f-482a-4ada-9f5d-934557eb22d1: output channel doesn't exist: 0
> 
> missing the 5V/10V selection
> 
> missing Name field for each Channel

but i think the problem lies in the pip-pi Lib witch is only written for the 8403

kizniche, 2024-03-09:
Did you set the right sda & scl pins when you tested the code?

silverhawk1983, 2024-03-09:
Tested Your Code in my dev system on Raspi 5  but like i sad no error and no output  but like i said  the 8413 not the 8403
[image]
[image]

[Nithalik](https://github.com/Nithalik) has the 8403 

did you look in the lib you used

silverhawk1983, 2024-03-09:
> Did you set the right sda & scl pins when you tested the code?

I2C ist working fine  with all other modules i have 

61 is SCD30 
5F is GP8413
5A is MLX  not shoing in overview some how    both the scd30 and the Mlx workling fine

kizniche, 2024-03-09:
>Rasp 5

SInce the library uses Rpi.GPIO, it is not compatible with Pi >= 5, as indicated in the Output module:

https://github.com/kizniche/Mycodo/blob/fd837e55fac4c604fa3c3ffa06caf566a801792e/mycodo/outputs/value_dac_p8403.py#L40

>I2C ist working fine with all other modules i have is set to gpio 2/3

THis doesn't directly answer the question of whether you set the same pins with the example code you posted. Also, see the note above about any library using RPi.GPIO, you need to use a Pi <= 4.

silverhawk1983, 2024-03-09:
that statement was not in the first try 2 days ago     i re downloaded today the module for the test and the pictures 
the only other pi i have is an old zero   thats on full throttle with mycodo not realy usefull

so it would be wise to get the original DFRobot Lib ubgraded for the pypi and without RPi.GPIO 
[https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi](https://github.com/DFRobot/DFRobot_GP8XXX/tree/master/python/raspberryPi )

but my python skills are not at a point to take that on   im happy that i can read and half way understand whats written there

Joe2824, 2024-03-30:
@silverhawk1983 I will try to update the pypi repository.
The problem is that the chip is using a non i2c standard to save values to the chip. Or I have to delve further into the smbus2 library.
I will try to adopt the DFRobot_GP8XXX. But right now I'm a limited with my time.

silverhawk1983, 2024-04-02:
@Joe2824 yeah   happy   to hear   i will test  the GP8413  as soon as posible  Thanks for the invest!
@kizniche  here is Joes new Lib [https://pypi.org/project/GP8XXX-IIC/](https://pypi.org/project/GP8XXX-IIC/)
hope that helps with a more universal implemention  so we can test the three modules that @Nithalik and i have on hand!

kizniche, 2024-04-02:
I just added a new Output module to test.

Joe2824, 2024-04-03:
@kizniche perfect thank you.
I found some bugs in my module and also in your code.
I will start a pull request when I'm done.

Is it possible to hide the channel1 when I add a custom option like this:
```python
'custom_options': [
        {
            'id': 'chip_select',
            'type': 'select',
            'options_select': [
                ('0', 'GP8503'),
                ('1', 'GP8211S'),
                ('2', 'GP8512'),
                ('3', 'GP8413'),
                ('4', 'GP8403'),
                ('5', 'GP8302'),
            ],
            'name': 'Device',
            'phrase': 'Select your GP8XXX device'
        }
    ],
```

kizniche, 2024-04-03:
I haven't done anything except copy the other closely-related module and add the code to use only GP8413. The number of channels is hard-coded in the module, so a different module will need to be created for any device variant that has a different number of channels.

Joe2824, 2024-04-10:
@silverhawk1983 I created a pull request #1375.
Could you help me to test?

silverhawk1983, 2024-04-11:
@Joe2824 sure   i ty your lib on the weekend   with the 8413 and some example code 
for the 8403  i cant test  i dont have that one

Joe2824, 2024-04-11:
@silverhawk1983 no worries I have the 8403 and this worked as intended.

silverhawk1983, 2024-04-14:
tested the  8413 just seams to work getting output    testet with your example file on RPI5   

only the autorange in your expample dont work

Joe2824, 2024-04-14:
@silverhawk1983 I made a mistake with the GP8431 data sheet and thought it only supported 0-10V. v0.0.4 should now have solved the problem.

silverhawk1983, 2024-04-14:
did it can be  that the channels are switched around channel 0 gives at Out2 out and  channel 1 at Out1

Joe2824, 2024-04-15:
You have three options 0 for channel 0, 1 for channel 1 and 2 for channel 0 & 1.

silverhawk1983, 2024-04-15:
sorry all fine   my wires did have a twist in it :)     allfine works good and as it should

silverhawk1983, 2024-04-17:
@Joe2824 did you could test the 8403 with mycodo did all works!

@kizniche for the Module would be nice to set names for each channel!

Joe2824, 2024-04-17:
@silverhawk1983 Yes, my 8403 has been running for about 5 days without any errors in production.

closed 2024-10-04 by kizniche