kizniche/Mycodo #1356: PCF8575 activating mutliple /wrong outputs on PCF8575 16 Channel Relay module

mirukisan, 2023-12-16:
Hello and thanks for the great software. 

I have an issue with a 16 Channel I2C relay module based on the PCF8575 chip running with Mycodo...kind of two issues: 
1. Outputs seem reversed. 
When turning on CH0: Channel 1 in Mycodo >>> Relay number 16 on the module is activated, 
When turning on CH1: Channel 2 in Mycodo >>> Relay number 15 on the module is activated,
...
...
When turning on CH14: Channel 15 in Mycodo >>> Relay number 2 on the module is activated,
When turning on CH15: Channel 16 in Mycodo >>> Relay number 1 on the module is activated.
I can live with that as long as the 2nd issue could be solved..

2. When turning in Mycodo any channel (0-15) output on, on the relay module relays 9 through 16 always turns on.
I can turn them individually off, but when turning any of the outputs on, relays 9 through 16 turn on again.

I have made a short video showing the issue which can be downloaded from:
[video link here](https://drive.google.com/file/d/10TqESFOIppCVO4bcEC6yTDNTQL3DblX8/view?usp=sharing) 

mycodo.log from the actions in the video: 

2023-12-16 12:48:42,935 - INFO - mycodo - Mycodo daemon started in 7.269 seconds
2023-12-16 12:50:20,048 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(on, 3, sec, 0.0, 0.0, True)
2023-12-16 12:50:20,054 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH3 (Name) ON at 2023-12-16 12:50:20.053322. Output returned: success
2023-12-16 12:50:23,338 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 3, None, 0.0, 0.0, True)
2023-12-16 12:50:23,340 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH3 (Name) OFF at 2023-12-16 12:50:23. Output returned: success
2023-12-16 12:50:23,761 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 2, None, 0.0, 0.0, True)
2023-12-16 12:50:23,764 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH2 (Name) OFF at 2023-12-16 12:50:23. Output returned: success
2023-12-16 12:50:24,331 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 1, None, 0.0, 0.0, True)
2023-12-16 12:50:24,333 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH1 (Name) OFF at 2023-12-16 12:50:24. Output returned: success
2023-12-16 12:50:24,779 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 0, None, 0.0, 0.0, True)
2023-12-16 12:50:24,781 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH0 (Name) OFF at 2023-12-16 12:50:24. Output returned: success
2023-12-16 12:50:25,592 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 4, None, 0.0, 0.0, True)
2023-12-16 12:50:25,594 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH4 (Name) OFF at 2023-12-16 12:50:25. Output returned: success
2023-12-16 12:50:26,895 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 5, None, 0.0, 0.0, True)
2023-12-16 12:50:26,898 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH5 (Name) OFF at 2023-12-16 12:50:26. Output returned: success
2023-12-16 12:50:27,843 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 6, None, 0.0, 0.0, True)
2023-12-16 12:50:27,845 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH6 (Name) OFF at 2023-12-16 12:50:27. Output returned: success
2023-12-16 12:50:28,230 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 7, None, 0.0, 0.0, True)
2023-12-16 12:50:28,233 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH7 (Name) OFF at 2023-12-16 12:50:28. Output returned: success
2023-12-16 12:50:30,953 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(on, 11, sec, 0.0, 0.0, True)
2023-12-16 12:50:30,955 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH11 (Name) ON at 2023-12-16 12:50:30.953950. Output returned: success
2023-12-16 12:50:32,813 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(on, 12, sec, 0.0, 0.0, True)
2023-12-16 12:50:32,815 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH12 (Name) ON at 2023-12-16 12:50:32.814375. Output returned: success
2023-12-16 12:50:34,353 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 12, None, 0.0, 0.0, True)
2023-12-16 12:50:34,356 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH12 (Name) OFF at 2023-12-16 12:50:34. Output returned: success
2023-12-16 12:50:36,355 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 11, None, 0.0, 0.0, True)
2023-12-16 12:50:36,357 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH11 (Name) OFF at 2023-12-16 12:50:36. Output returned: success
2023-12-16 12:50:38,281 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 0, None, 0.0, 0.0, True)
2023-12-16 12:50:38,283 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH0 (Name) OFF at 2023-12-16 12:50:38. Output returned: success
2023-12-16 12:50:38,833 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 1, None, 0.0, 0.0, True)
2023-12-16 12:50:38,839 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH1 (Name) OFF at 2023-12-16 12:50:38. Output returned: success
2023-12-16 12:50:39,659 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 2, None, 0.0, 0.0, True)
2023-12-16 12:50:39,661 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH2 (Name) OFF at 2023-12-16 12:50:39. Output returned: success
2023-12-16 12:50:40,446 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 3, None, 0.0, 0.0, True)
2023-12-16 12:50:40,448 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH3 (Name) OFF at 2023-12-16 12:50:40. Output returned: success
2023-12-16 12:50:41,285 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 4, None, 0.0, 0.0, True)
2023-12-16 12:50:41,287 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH4 (Name) OFF at 2023-12-16 12:50:41. Output returned: success
2023-12-16 12:50:42,349 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 5, None, 0.0, 0.0, True)
2023-12-16 12:50:42,352 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH5 (Name) OFF at 2023-12-16 12:50:42. Output returned: success
2023-12-16 12:50:43,407 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 6, None, 0.0, 0.0, True)
2023-12-16 12:50:43,410 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH6 (Name) OFF at 2023-12-16 12:50:43. Output returned: success
2023-12-16 12:50:44,297 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 7, None, 0.0, 0.0, True)
2023-12-16 12:50:44,298 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH7 (Name) OFF at 2023-12-16 12:50:44. Output returned: success
2023-12-16 12:50:47,379 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(on, 4, sec, 0.0, 0.0, True)
2023-12-16 12:50:47,380 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH4 (Name) ON at 2023-12-16 12:50:47.379547. Output returned: success
2023-12-16 12:50:49,895 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 4, None, 0.0, 0.0, True)
2023-12-16 12:50:49,897 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH4 (Name) OFF at 2023-12-16 12:50:49. Output returned: success
2023-12-16 12:50:52,220 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - output_on_off(off, 6, None, 0.0, 0.0, True)
2023-12-16 12:50:52,222 - DEBUG - mycodo.outputs.on_off_pcf8575_48e3604c - Output 48e3604c-7310-42b7-9980-6e679c816324 CH6 (Name) OFF at 2023-12-16 12:50:52. Output returned: success
[... cut ...]
Frontend Process ID: 564
Frontend RAM Usage: 86.824 MB
Frontend Virtualenv: Yes

uname -r
6.1.0-rpi7-rpi-v7

cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
 
Please let me know how I can assist solving the issue. 
Thank you

kizniche, 2023-12-16:
Have you verified this is happening on the actual pcf8575 chip? I can't be responsible for every piece of hardware that is wired different ways to the IC or code that's designed to interface with those specific hardwares.

mirukisan, 2023-12-16:
Hi, unfortunately the issue is happening also with cheap PCF8575TS modules brought from aliexpress, link of the modules an be found [here](https://www.aliexpress.com/item/1005001953422378.html)
A close up picture of the module can be found [here](https://ae01.alicdn.com/kf/Hbc36d57d6d484661b50ed92735f945ccB/diymore-PCF8575TS-Expansion-Module-I2C-Communication-Control-Interface-to-16-I-O-Interface-Converter-Board-for.jpg_.webp)
I have 3 of those, same behavior with all 3... 
Video proof can be found [here](https://drive.google.com/file/d/10VG37MGZ8Ux7hWB8_WMHSmb-VCqRWuEe/view?usp=sharing)
Thank you

kizniche, 2023-12-19:
We can't edit the current Output Modules since this may cause issues with current users of them, but we can create a new output module. If you have working Python code, have you attempted to duplicate the current Output Module and replace the code that communicates with the device with the working code?

mirukisan, 2023-12-22:
Since I don't have programming skills I've tried the last days to modify the code using chatgpt...
I tried to modify the PCF8575 [code from here](https://github.com/kizniche/Mycodo/blob/master/mycodo/outputs/on_off_pcf8575.py) After encountering many indentation issues and many more failed attempts to get the outputs work correctly I will have to give up... I am sorry. 
If possible could you please take a look at the code and make some changes to solve the 2nd issue (where outputs 10- 17 always turn on). I am more than happy to test the code if available. 
Thank you and Happy Holidays!

kizniche, 2023-12-22:
Can you link to code that works?

mirukisan, 2023-12-23:
Unfortunately no, I don't have any code that works.

mirukisan, 2023-12-26:
Hi, finally some good news, I think I've found the issue. In the original code from /mycodo/outputs/on_off_pcf8575.py there is this line under the class PCF8575(object):
        new_state = current_state | bit if value else current_state & (~bit & 0xff)
I've removed the last part (0xff) so it looks like this:
        new_state = current_state | bit if value else current_state & ~bit 
and now it works as expected.  
As per chatgpt: "The 0xff is used to ensure that only the lower 8 bits are considered in the negation operation (~bit). This is often used when working with byte-sized bitmasks to avoid unexpected behavior with higher bits."
Please if someone could check if I don't get some unexpected behavior by leaving this so that would be great.
[pcf8575ok.txt](https://github.com/kizniche/Mycodo/files/13773987/pcf8575ok.txt)

As for the other issue with the reversed outputs, well they are still revered, but at least on the correct side.
With the reversed outputs issue I mean now:
when turning CH0: Channel 1 in Mycodo, on the PCF8575 channel P07 turns on
when turning CH1: Channel 2 in Mycodo, on the PCF8575 channel P06 turns on
and so on.
when turning CH8: Channel 9 in Mycodo, on the PCF8575 channel P17 turns on
when turning CH9: Channel 10 in Mycodo, on the PCF8575 channel P16 turns on
but I can leave happy with that.
[pcf8575ok_reversed.txt](https://github.com/kizniche/Mycodo/files/13774022/pcf8575ok_reversed.txt)
Attached the code for reference.
Thank you

mirukisan, 2024-03-10:
Hi, unfortunately the issue is still not solved, with the attached code all outputs except "CH5: Channel 6" and "CH6: Channel 7" are working, however the mentioned channels cannot be controlled individually, when turning a channel on, both turn on, when turning any of the above mentioned channels off, both turn off. Please help/advise, Thank you.
[pcf8575_ch_issues.txt](https://github.com/kizniche/Mycodo/files/14551871/pcf8575_ch_issues.txt)

mirukisan, 2024-03-10:
never mind seems the PCF8575 chip is broken, same behavior with arduino code...

mirukisan, 2024-04-26:
Just an update/FYI:  I've solved the two issues described in my first post by replacing this line under the class PCF8575(object):
new_state = current_state | bit if value else current_state & (~bit & 0xff)
with this line:
new_state = current_state & ~(1 << output_number) if not value else current_state | (1 << output_number)

closed 2024-04-26 by mirukisan

kizniche, 2024-04-26:
I wouldn't be surprised if this was a counterfeit chip. You can have a lot of seemingly aberrant behavior with counterfeit chips. Why it's often better to go with a reputable vendor. Not saying it's definitely the case here, but this module was initially tested and worked as expected. Nothing about the module has changed since then.