kizniche/Mycodo #1358: Zigbee2MQTT Plugin for Mycodo

FridgeGrow, 2023-12-23:
**Is your feature request related to a problem? Please describe.**

Yes, the feature request is related to a problem. Currently, it is not possible to integrate Zigbee devices directly into Mycodo. To do this, a Mosquitto Broker as well as Zigbee2Mqtt must be installed on the Raspberry Pi. This is an additional and more complex task that not all users want to do. 

**Describe the solution you'd like**

I would like a Zigbee2Mqtt plugin for Mycodo. This plugin would make it possible to integrate Zigbee devices directly into Mycodo. This would offer users a number of benefits, including:

Easier configuration: Zigbee2Mqtt provides a simple and intuitive user interface for configuring Zigbee devices.
More flexible control: Zigbee2Mqtt offers a variety of ways to control Zigbee devices, such as through rules and scenes.
Improved security: Zigbee2Mqtt offers a number of security features to help ensure the security of Zigbee devices.

**Describe alternatives you've considered**

The only alternative I have considered is the method , which requires installing Mosquitto Broker and Zigbee2Mqtt on the Raspberry Pi. I use this Setup with a Sonoff Dongle 3.0 and controll for example a Ikea tradfry smart plug however, this method is more complex and not suitable for all users.

**Additional context**

Zigbee2Mqtt is an open source project that allows Zigbee devices to be connected to MQTT protocols. MQTT is a lightweight protocol that is ideal for connecting IoT devices.

Zigbee2Mqtt supports a wide variety of Zigbee devices, including sensors, actuators, and smart home devices.

A Zigbee2Mqtt plugin for Mycodo would extend the functionality of Mycodo and make it possible to integrate a wider range of Zigbee devices. This would make the system more attractive to more users.

kizniche, 2023-12-23:
Can you provide more information? Software link(s), example format, how this broker differs from Mosquito, etc.

FridgeGrow, 2023-12-25:
The following software is installed on my Raspberry Pi 4(4GB):

Raspberry Pi OS (64-bit)
mosquitto MQTT broker
Zigbee2MQTT (https://github.com/Koenkk/zigbee2mqtt) (https://www.zigbee2mqtt.io/)
SONOFF Zigbee Gateway, ZBDongle-E 3.0 USB
I was looking for a way to integrate Smart Plugs into mycodo.
In Switzerland, the Smart Plug selection is different. Thanks to the Zigbee standard, I have the option to integrate 2 Smart Plugs:

Phillips Hue Smart Plug (CH)
IKEA Tradfry Smart Plug (CH)

Client → Mycodo sends OUTPUT [IP] On/Off: MQTT Publish (paho-mqtt)
Broker→ Mosquitto provides data
Client → Zigbee2MQTT forwards commands to devices

I am an absolute beginner and I can't give you any more information about this.
Also not whether it is technically feasible.

Here is the list of compatible devices:
https://www.zigbee2mqtt.io/supported-devices/

I am also still learning more about this myself. The next step is to integrate a sensor via Zigbee2MQTT.

For me, the main goal was to be able to use Smart Plugs and with the connected IKEA Tradfry Smart Plug, the control works flawlessly.

kizniche, 2024-01-11:
Just had some new things to play with arrive...

[image]

kizniche, 2024-01-12:
Zigbee2MQTT was pretty straightforward, all devices were detected and automatically configured on the first connect attempt.

[image]

[image]