kizniche/Mycodo #1359: Bang-Bang controller hangs randomly.  Can not acquire a measurement, but all inputs and calculation Functions are working with no errors.

LucidEye, 2024-01-02:
This is part of the ongoing problem I have been having for the past 8 months with Mycodo hanging or becoming un-responsive.

I just upgraded to Mycodo 8.15.13 6 days ago.
Running on a Pi 3B+ - 64-bit Bullseye OS - InfluxDB v2

I am still seeing my VPD humidistat Bang-Bang Function hanging or becoming un-responsive.

For no apparent reason, this controller will suddenly stop working and I get the following errors in the log...

2024-01-01 15:55:55,626 - ERROR - mycodo.function.bang_bang_3397e924 - Could not acquire a measurement
2024-01-01 15:56:02,448 - ERROR - mycodo.function.bang_bang_3397e924 - Could not acquire a measurement
2024-01-01 15:56:09,470 - ERROR - mycodo.function.bang_bang_3397e924 - Could not acquire a measurement
2024-01-01 15:56:17,174 - ERROR - mycodo.function.bang_bang_3397e924 - Could not acquire a measurement
2024-01-01 15:56:24,335 - ERROR - mycodo.function.bang_bang_3397e924 - Could not acquire a measurement

I don't understand why it can't acquire a measurement, all of my Inputs and calculation Functions are working fine and not throwing any errors in the logs.
The process I'm using to get the measurement for that bang-bang controller is as follows...

Input - SHTC3 temp/humidity sensor - working fine, acquiring temp and humidity measurements and giving no errors in the log.
[image]

I then take that air Temperature reading and perform an Equation: Single-Measure Function to subtract 5 degrees F.  This adjusted temperature is now the approximate average leaf temperature. (I am getting no errors in the log from this Function).
[image]

Next, I take that approximate leaf temperature, and the relative humidity reading from the SHTC3 Input, and perform a VPD Function calculation to get the approximate actual leaf VPD.  Again, I am getting no errors in the log from this Function, and the live page and dashboards show this measurement with no problems.
[image]
 
The Leaf VPD calculated from the previous function is what the humidistat bang-bang is using to control VPD in the grow space... so somewhere in between the VPD calculation function and the Bang Bang controller, Mycodo seems to be 'dropping the ball' in keeping track of this measurement, for reasons I still can not determine.  

The only way I know the Bang-Bang controller has actually failed is because of the Conditional Controller I have running that makes an LED blink on the status board I built.  I programmed it to blink blue if for some reason the Bang Bang had no measurement....
[image]

This is the Conditional code...
blinks=8
measurement = self.condition("1c8c49f3") #VPD [Function24] Status LED Conditional
if measurement is not None:  # If a measurement exists
    if measurement < 0.8:  # led is dark blue
        self.run_action("1a811689", value={"payload": "0,0,2"})
    elif measurement >= 0.8 and measurement < 1.3:  # led is green
        self.run_action("1a811689", value={"payload": "0,1,0"})
    elif measurement >= 1.3 and measurement < 1.45:  # led is yellow
        self.run_action("1a811689", value={"payload": "1,1,0"})
    elif measurement >= 1.45:  # led is bright red
        self.run_action("1a811689", value={"payload": "2,0,0"})
else:  # If no measurement exists led is flashing bright blue
    for _ in range(blinks):
        self.run_action("1a811689", value={"payload": "0,0,10"})
        sleep(.75)
        self.run_action("1a811689", value={"payload": "0,0,0"})
        sleep(.75)

If I see my status board LED blinking blue, and I catch it in time, I can open the web-gui and restart the VPD humidistat Bang-Bang, which seems to temporarily resolve the error.  However, if I let the error state run too long, the system will eventually become un-responsive and I lose any access to the GUI or even an SSH session and the system requires a hard reboot to fix this problem.... but again it is only temporary... this error keeps happening randomly every few days.

kizniche, 2024-01-03:
You're likely running into system resource allocation issues from so many things running (judging by your number of inputs (29+) and functions (27+) in the screenshot). Have you considered consolidating, such as using the Equation Input Action instead of an entire Function to modify the value?

LucidEye, 2024-01-04:
I do not have 29 inputs.... I have 13 inputs currently setup, and only 4 of them are activated.
[image]

I do have a lot of functions, but again, less than half of them are active at any given time (15-20), and most of them are simple Trigger: Daily Time Point Functions, so I can't see how those would be using up system resources.

And even if that were the case, I would expect to see problems with all of the Inputs and Functions, not just this one Bang-Bang constantly failing.

Regardless, all of this used to work just fine with no crashing, I haven't made any major changes to my system setup in about 2 years, but all of this started after upgrading from 8.15.7 to 8.15.8 and it hasn't stopped.

The Equation Input Action will not give me 2 measurements... it will only give me one as it takes the original measurement from the sensor, performs the equation, and then writes ONLY that new adjusted measurement into the database.  I need BOTH the air temp (original measurement), AND the -5 adjusted measurement for approximate leaf temp so I can calculate correct leaf VPD (not room VPD).

LucidEye, 2024-07-01:
The only way I was able to "fix" this issue was to run Mycodo on a Pi 4 and buy an SSD to use as the system drive.
Apparently what was happening was an I/O bottleneck.  
The Pi was doing it's data acquisition and calculations, but the SD card write speed was slower than the Pi's computing speed.
So what was essentially happening was read/write lag... the Pi could not find the measurement because that measurement was still sitting in the I/O buffer waiting to be written to the SD card.
Using an SSD with a high write speed, plugged into the Pi 4's USB3 port, is the only way to avoid this read/write I/O bottleneck.
I think this issue can finally be closed.