kizniche/Mycodo #1360: issue with TCA9548a multiplexer

Nithalik, 2024-01-11:
I am using a dfrobot multiplexer

https://wiki.dfrobot.com/Gravity__Digital_1-to-8_I2C_Multiplexer_SKU_DFR0576#target_5

uses the TCA9548a chip. 

i went to enable it given your instructions at https://kizniche.github.io/Mycodo/I2C-Multiplexers/

when enabled it now shows 2 i2c channels oddly, screenshot below

[image]

the 11 and 12 channel will not go away and nothing in those two channels is being recognized by the system.

now the odd part. when i have the multiplexer plugged in, it shows all other channels normally, but still has those two filled like that, when unplugged just like the screenshot

i went to the boot config file and removed the multiplex line - these two channels still show even after rebooting system

i reinstalled mycodo after - these two channels still show.

it basically does not matter what the condition now, it still shows those two channels

### Versions:

 - Mycodo Version: 8.15.13
 - Raspberry Pi Version: *[e.g. 5]*
 - Raspbian OS Version: *raspian full*

### Reproducibility

really not sure how to reproduce here, i put the pi back in its "base" state to the channels still showing

### Expected behavior

the two i2c channels should not be populating randomly

kizniche, 2024-01-22:
You haven't provided enough information to know if you set it up correctly. The bug report template instructs to provide detailed step-by-step instructions for how to replicate the issue. This is the most crucial piece of information you can provide, but it appears to be omitted from your bug report.

Nithalik, 2024-02-07:
I have received a new RPI5, i really didnt know how to show reproduction steps before.

1. Install RPI OS on brand new RPI 5, nothing every connected to GPIO
2. Install mycodo using curl -L https://kizniche.github.io/Mycodo/install | bash
3. go to system information and the below pic shows

kizniche, 2024-02-07:
So you haven't enabled I2C and i2cdetect is showing you have active I2C busses?

Nithalik, 2024-02-07:
you might have seen the un-updated comment.

i2c is enabled, was the only thing i did once installed the os

dgedgedge, 2024-02-17:
Please consider the following issue on RapiOS :
https://github.com/raspberrypi/linux/issues/5914

Nithalik, 2024-04-30:
Hi,

Just wanted to follow up here, all moved into my new place and starting up my mycodo systems again.

is there no support for the RPI5? this seems like its related to the GPIO issues the RPI5 is having.

I ended up with another new RPI5 and am getting the same thing as previous comment