kizniche/Mycodo #1361: Python 3 code(v2.0) not showing output

zakman1805, 2024-01-12:
Hello there,

thank you for building this wonderful software.

I ran some Python API code to get, the status of a device from Tuya APi, which works very well using Python on terminals. However I tried the same code on Mycodo Python code(v2.0), it works but was on able to display the temperature while leaving other parameters out. I don't know why this is happening, the temperature output that works was parsed from the same json where other parameters are.
below is my code:

from tuya_connector import TuyaOpenAPI
import json
# Cloud project authorization info	
ACCESS_ID = 'id'	
ACCESS_KEY = 'key'
ENDPOINT = "https://openapi.tuyaeu.com/"	
		

# Initialization of tuya openapi	
openapi = TuyaOpenAPI(ENDPOINT, ACCESS_ID, ACCESS_KEY)	
openapi.connect()#USERNAME, PASSWORD)

request = openapi.get('/v1.0/iot-03/devices/deviceid/status')	
Decode = json.dumps(request)
Output = json.loads(Decode)
w = Output['result']
Humidity = (w[3])
Temperature = (w[4])
Power = (w[0])
Dehumidify = (w[1])
Fan_speed = (w[2])
return{0,Dehumidify['value']} 

the returned value does not display

so I tested with the default code that generates a random number from 0-100 which works
then I adjusted the code like this:
import random  # Import any external libraries

# Get measurements/values (for example, these are randomly-generated numbers)
random_value = 100

# Store measurements in database (must return a dictionary of channels and values)
return {0: random_value}
this fail to read data the 100 I assigned

could you please look into this

thank you

kizniche, 2024-01-12:
Please indicate what version Mycodo you are using and reformat your code so it can be understood.

zakman1805, 2024-01-12:
[image]
[image]
[image]
[image]
[image]

I have attached the picture
 Mycodo version is 8.15.13
I ran the default code (random number between (10-100) which runs perfectly

zakman1805, 2024-01-12:
When I assigned 500 to the variable
random_value
i cant see any data again
And this happened to my python code too.
I want want us to look why the value is nothing after changing the value to 500

kizniche, 2024-01-12:
Once a type is assigned in influxdb, you must always use that type for that measurement. You are trying to store an integer after initially storing a float.

zakman1805, 2024-01-12:
Ohh, i get it 
Thank you. I does solve it
ill play around it and see if I see any issues else ill close the comment
thank you once again

zakman1805, 2024-01-15:
Hey Kizniche,
thank you for your time, well appreciated.
I played around with the Python inputs, and I ran into the same issues. I tried several Python input with the same code just to make sure I initial a float variable for the influx DB but only two values which was pre-defined was was shown.
[image]
However, I can testify that the code output a float result when i tested via Geany 
[image]
what do you think might cause this?

zakman1805, 2024-01-15:
Here's my code
from tuya_connector import TuyaOpenAPI
import json
# Cloud project authorization info	
ACCESS_ID = 'rvprfe8'	
ACCESS_KEY = '4c3c4eb9d06'
ENDPOINT = "https://openapi.tuyaeu.com/"	

# Initialization of tuya openapi	
openapi = TuyaOpenAPI(ENDPOINT, ACCESS_ID, ACCESS_KEY)	
openapi.connect()
request = openapi.get('/v1.0/iot-03/devices/430393svd2/status')	
Decode = json.dumps(request)
Output = json.loads(Decode)
w = Output['result']
Power = (w[0])
Dehumidify = (w[1])
Dehumidify1 = float(int(Dehumidify['value']))
Fan_speed = (w[2])
Humidity = (w[3])
Humidity1 = float(int(Humidity['value']))
Temperature = (w[4])
Temperature1 = float(int(Temperature['value']))
if Power['value'] == "True":
	Power = 1.0
else:
	Power = 0.0
if Fan_speed['value'] == "high":
	Fan_speed = 1.0
else:
	Fan_speed = 0.0

return {0: Power,1: Dehumidify1,2: Fan_speed,3: Humidity1,4: Temperature1}

kizniche, 2024-01-15:
Are you sure you didn't previously store a different data type for that measurement unit? You can delete your influxdb from the Diagnostic menu and start over. This is something you'll need to read the influxdb docs to understand.

zakman1805, 2024-01-16:
Okay , could you share the link/document, scare to delete it. let me read the document

zakman1805, 2024-01-18:
Thank you everything works perfect

closed 2024-01-18 by zakman1805

DanWatters1, 2024-03-21:
Hi @zakman1805, I am intersted in switching from KASA KP303 which is no longer available to the Tuya power board. I am not a programmer, just a hobby grower using Rasberry Pi and the Mycodo program which is brilliant. I was wondering if I could get someone who would help me set up the tuya power board on the Mycodo platform. Is that what you've done using Python? Could I pinch your code and set this up on my system, please?