kizniche/Mycodo #1362: Blank Widgets

Frosty-Burrito, 2024-01-18:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

The dashboard does not display any widgets properly. This occurred after updating to the latest version of mycodo. Creating new dashboards and adding new widgets does not make a difference. The widget size can be changed, they can be dragged around, the setting button does work, however using any of that functionality does not make the widgets display any information.

### Versions:

 - Mycodo Version: 8.15.13
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Bookworm

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Open Dashboard
2. See Error

### Expected behavior

Dashboard widgets should display properly

### Screenshots

My normal dashboard, blank:
[image]

New test dashboard, also blank:
[image]

### Additional context

I have tried viewing the dashboard din firefox and chrome, I get the same results in both.

kizniche, 2024-01-18:
Have you tried creating a new dashboard?

Frosty-Burrito, 2024-01-18:
Yes, doesn't change anything

kizniche, 2024-01-18:
What does inspecting the page (console) yield?

kizniche, 2024-01-18:
Have you inspected the upgrade log for errors?

Frosty-Burrito, 2024-01-18:
[image]
That's what I see

Frosty-Burrito, 2024-01-18:
Here are the last 200 lines of the upgrade log:
`[2024-01-16 12:26:03] python3-setuptools is already the newest version (66.1.1-1).
[2024-01-16 12:26:03] python3-venv is already the newest version (3.11.2-1+b1).
[2024-01-16 12:26:03] rng-tools-debian is already the newest version (2.3).
[2024-01-16 12:26:03] sqlite3 is already the newest version (3.40.1-2).
[2024-01-16 12:26:03] unzip is already the newest version (6.0-28).
[2024-01-16 12:26:03] wget is already the newest version (1.21.3-1+b1).
[2024-01-16 12:26:03] The following packages were automatically installed and are no longer required:
[2024-01-16 12:26:03]   gtk2-engines-clearlookspix libpisp0.0.1 linux-headers-6.1.0-rpi4-common-rpi
[2024-01-16 12:26:03]   linux-headers-6.1.0-rpi4-rpi-2712 linux-headers-6.1.0-rpi4-rpi-v8
[2024-01-16 12:26:03]   linux-image-6.1.0-rpi4-rpi-2712 linux-image-6.1.0-rpi4-rpi-v8 tightvncpasswd
[2024-01-16 12:26:03] Use 'apt autoremove' to remove them.
[2024-01-16 12:26:03] 0 upgraded, 0 newly installed, 0 to remove and 47 not upgraded.
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] #### Installing and configuring nginx web server
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] #### Disabling service for nginx web server
[2024-01-16 12:26:03] Removed "/etc/systemd/system/mycodoflask.service".
[2024-01-16 12:26:03] Removed "/etc/systemd/system/multi-user.target.wants/mycodoflask.service".
[2024-01-16 12:26:05] 
[2024-01-16 12:26:05] #### Enabling service for nginx web server
[2024-01-16 12:26:05] Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
[2024-01-16 12:26:05] Executing: /lib/systemd/systemd-sysv-install enable nginx
[2024-01-16 12:26:11] Created symlink /etc/systemd/system/mycodoflask.service â /home/frosty-burrito/Mycodo/install/mycodoflask.service.
[2024-01-16 12:26:11] Created symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service â /home/frosty-burrito/Mycodo/install/mycodoflask.service.
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Installing logrotate scripts
[2024-01-16 12:26:13] Mycodo logrotate script installed
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Checking Python 3 virtual environment
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Updating pip
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Checking Python 3 virtual environment
[2024-01-16 12:26:14] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2024-01-16 12:26:14] Requirement already satisfied: pip in ./env/lib/python3.11/site-packages (23.3.1)
[2024-01-16 12:26:16] Collecting pip
[2024-01-16 12:26:16]   Downloading https://www.piwheels.org/simple/pip/pip-23.3.2-py3-none-any.whl (2.1 MB)
[2024-01-16 12:26:16]      ââââââââââââââââââââââââââââââââââââââââ 2.1/2.1 MB 3.1 MB/s eta 0:00:00
[2024-01-16 12:26:18] Installing collected packages: pip
[2024-01-16 12:26:18]   Attempting uninstall: pip
[2024-01-16 12:26:18]     Found existing installation: pip 23.3.1
[2024-01-16 12:26:19]     Uninstalling pip-23.3.1:
[2024-01-16 12:26:19]       Successfully uninstalled pip-23.3.1
[2024-01-16 12:26:22] Successfully installed pip-23.3.2
[2024-01-16 12:26:24] 
[2024-01-16 12:26:24] #### Installing pip requirements
[2024-01-16 12:26:24] 
[2024-01-16 12:26:24] #### Checking Python 3 virtual environment
[2024-01-16 12:26:25] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2024-01-16 12:26:25] Requirement already satisfied: alembic==1.11.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 2)) (1.11.2)
[2024-01-16 12:26:26] Collecting argparse==1.4.0 (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 3))
[2024-01-16 12:26:26]   Using cached https://www.piwheels.org/simple/argparse/argparse-1.4.0-py2.py3-none-any.whl (23 kB)
[2024-01-16 12:26:26] Requirement already satisfied: bcrypt==3.2.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 4)) (3.2.2)
[2024-01-16 12:26:26] Requirement already satisfied: distro==1.8.0 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 5)) (1.8.0)
[2024-01-16 12:26:27] Requirement already satisfied: email_validator==1.3.1 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 6)) (1.3.1)
[2024-01-16 12:26:27] Requirement already satisfied: filelock==3.12.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 7)) (3.12.2)
[2024-01-16 12:26:27] Requirement already satisfied: Flask==2.3.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 8)) (2.3.2)
[2024-01-16 12:26:27] Requirement already satisfied: Flask_Accept==0.0.6 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 9)) (0.0.6)
[2024-01-16 12:26:27] Requirement already satisfied: Flask_Babel==3.1.0 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 10)) (3.1.0)
[... cut ...]
[2024-01-16 12:27:23] 2024-01-16 12:27:23,422 Pypi Package: adafruit-circuitpython-hcsr04, Required: v0.4.6, Installed: v0.4.6, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,591 Pypi Package: python-kasa, Required: v0.5.0, Installed: v0.5.0, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,593 Pypi Package: aio_msgpack_rpc, Required: v0.2.0, Installed: v0.2.0, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,761 Pypi Package: RPi.GPIO, Required: v0.7.1, Installed: v0.7.1, Same: True`

Frosty-Burrito, 2024-01-18:
last 200 lines of upgrade log:

> [2024-01-16 12:26:03] python3-setuptools is already the newest version (66.1.1-1).
[2024-01-16 12:26:03] python3-venv is already the newest version (3.11.2-1+b1).
[2024-01-16 12:26:03] rng-tools-debian is already the newest version (2.3).
[2024-01-16 12:26:03] sqlite3 is already the newest version (3.40.1-2).
[2024-01-16 12:26:03] unzip is already the newest version (6.0-28).
[2024-01-16 12:26:03] wget is already the newest version (1.21.3-1+b1).
[2024-01-16 12:26:03] The following packages were automatically installed and are no longer required:
[2024-01-16 12:26:03]   gtk2-engines-clearlookspix libpisp0.0.1 linux-headers-6.1.0-rpi4-common-rpi
[2024-01-16 12:26:03]   linux-headers-6.1.0-rpi4-rpi-2712 linux-headers-6.1.0-rpi4-rpi-v8
[2024-01-16 12:26:03]   linux-image-6.1.0-rpi4-rpi-2712 linux-image-6.1.0-rpi4-rpi-v8 tightvncpasswd
[2024-01-16 12:26:03] Use 'apt autoremove' to remove them.
[2024-01-16 12:26:03] 0 upgraded, 0 newly installed, 0 to remove and 47 not upgraded.
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] #### Installing and configuring nginx web server
[2024-01-16 12:26:03] 
[2024-01-16 12:26:03] #### Disabling service for nginx web server
[2024-01-16 12:26:03] Removed "/etc/systemd/system/mycodoflask.service".
[2024-01-16 12:26:03] Removed "/etc/systemd/system/multi-user.target.wants/mycodoflask.service".
[2024-01-16 12:26:05] 
[2024-01-16 12:26:05] #### Enabling service for nginx web server
[2024-01-16 12:26:05] Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
[2024-01-16 12:26:05] Executing: /lib/systemd/systemd-sysv-install enable nginx
[2024-01-16 12:26:11] Created symlink /etc/systemd/system/mycodoflask.service â /home/frosty-burrito/Mycodo/install/mycodoflask.service.
[2024-01-16 12:26:11] Created symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service â /home/frosty-burrito/Mycodo/install/mycodoflask.service.
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Installing logrotate scripts
[2024-01-16 12:26:13] Mycodo logrotate script installed
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Checking Python 3 virtual environment
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Updating pip
[2024-01-16 12:26:13] 
[2024-01-16 12:26:13] #### Checking Python 3 virtual environment
[2024-01-16 12:26:14] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2024-01-16 12:26:14] Requirement already satisfied: pip in ./env/lib/python3.11/site-packages (23.3.1)
[2024-01-16 12:26:16] Collecting pip
[2024-01-16 12:26:16]   Downloading https://www.piwheels.org/simple/pip/pip-23.3.2-py3-none-any.whl (2.1 MB)
[2024-01-16 12:26:16]      ââââââââââââââââââââââââââââââââââââââââ 2.1/2.1 MB 3.1 MB/s eta 0:00:00
[2024-01-16 12:26:18] Installing collected packages: pip
[2024-01-16 12:26:18]   Attempting uninstall: pip
[2024-01-16 12:26:18]     Found existing installation: pip 23.3.1
[2024-01-16 12:26:19]     Uninstalling pip-23.3.1:
[2024-01-16 12:26:19]       Successfully uninstalled pip-23.3.1
[2024-01-16 12:26:22] Successfully installed pip-23.3.2
[2024-01-16 12:26:24] 
[2024-01-16 12:26:24] #### Installing pip requirements
[2024-01-16 12:26:24] 
[2024-01-16 12:26:24] #### Checking Python 3 virtual environment
[2024-01-16 12:26:25] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2024-01-16 12:26:25] Requirement already satisfied: alembic==1.11.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 2)) (1.11.2)
[2024-01-16 12:26:26] Collecting argparse==1.4.0 (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 3))
[2024-01-16 12:26:26]   Using cached https://www.piwheels.org/simple/argparse/argparse-1.4.0-py2.py3-none-any.whl (23 kB)
[2024-01-16 12:26:26] Requirement already satisfied: bcrypt==3.2.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 4)) (3.2.2)
[2024-01-16 12:26:26] Requirement already satisfied: distro==1.8.0 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 5)) (1.8.0)
[2024-01-16 12:26:27] Requirement already satisfied: email_validator==1.3.1 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 6)) (1.3.1)
[2024-01-16 12:26:27] Requirement already satisfied: filelock==3.12.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 7)) (3.12.2)
[2024-01-16 12:26:27] Requirement already satisfied: Flask==2.3.2 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 8)) (2.3.2)
[2024-01-16 12:26:27] Requirement already satisfied: Flask_Accept==0.0.6 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 9)) (0.0.6)
[2024-01-16 12:26:27] Requirement already satisfied: Flask_Babel==3.1.0 in ./env/lib/python3.11/site-packages (from -r /home/frosty-burrito/Mycodo/install/requirements.txt (line 10)) (3.1.0)
[... cut ...]
[2024-01-16 12:27:23] 2024-01-16 12:27:23,422 Pypi Package: adafruit-circuitpython-hcsr04, Required: v0.4.6, Installed: v0.4.6, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,591 Pypi Package: python-kasa, Required: v0.5.0, Installed: v0.5.0, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,593 Pypi Package: aio_msgpack_rpc, Required: v0.2.0, Installed: v0.2.0, Same: True
[2024-01-16 12:27:23] 2024-01-16 12:27:23,761 Pypi Package: RPi.GPIO, Required: v0.7.1, Installed: v0.7.1, Same: True

kizniche, 2024-01-18:
You need to inspect the entire upgrade log for errors. The last 200 lines is not sufficient to determine if there was an issue.

Frosty-Burrito, 2024-01-18:
> You need to inspect the entire upgrade log for errors. The last 200 lines is not sufficient to determine if there was an issue.

I'll grab and upload it, hold on

kizniche, 2024-01-18:
If that's the last lines, your upgrade didn't complete. Run "sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh"

Frosty-Burrito, 2024-01-18:
And just like that all problems are solved, thank you.

closed 2024-01-18 by Frosty-Burrito