kizniche/Mycodo #1364: mycodo ERROR - mycodo.influx - Unknown Influxdb version:

danOS144, 2024-01-28:
Hi there,
I have a fresh installed Mycodo ( Mycodo 8.15.13 /influxDB v.1.8.10/ raspberry pi OS lite (32 bit) on a Raspberry zero W)
For testing i added the cpu-Load input and activated it. When I go to the live-Measurement page the input is shown but it gets no data.
The mycodo-logs show: 2024-01-28 11:58:35,001 - ERROR - mycodo.influx - Unknown Influxdb version: 
So I guess the problem is Mycodo can't connect to influxDB. 
I would be happy if anyone knows a solution for this, or can give me advice how to handle this! Thankful for help

Dan

### Versions:

 - Mycodo Version: *[8.15.13]*
 - Raspberry Pi Version: *[Raspberry zero W]*
 - Raspbian OS Version: *[Raspberry Pi OS lite (32 bit)]*

silverhawk1983, 2024-02-14:
it installs the latest version 2.0 of influxdb  but wants only older 1.8 and drops out

danOS144, 2024-02-14:
this threat is solved in :
https://forum.radicaldiy.com/t/error-mycodo-influx-unknown-influxdb-version/1773

the error occured because mycodo didn't install Influxdb at all....

kizniche, 2024-02-14:
It will attempt to install whichever version of influxdb you specify. Whether or not you have a 64-bit CPU and OS to be able to install the 64-bit version of the software (2.x) is another issue that is dependent on the user (user error). You must be knowledgeable of your CPU and OS to know if you can even install v2.x.

closed 2024-02-14 by kizniche