kizniche/Mycodo #1372: PID widgets display null instead of negative values

drgrumpy, 2024-03-14:
Recently upgraded from 8.4 to 8.15.3, via clean install on new SD card.

Seems that PID widgets display NULL instead of negative values: see screenshot:

[image]

Been checking for a couple of days - tt seems the PID controls shown are both running as expected (as on 8.4 running for years), i.e. at time of screenshot the cooiing PID is negative and output is active,  so purely a display issue. When values are zero or positive they display as expected. I also notice also that the PID widgets no longer display a graph of the values, so presumably there has been rewrite somewhere, but I couldn't spot in the changelog.

drgrumpy, 2024-03-14:
Also may be related, the Cooling PWM output duty cycle % does not display on the main graph, despite being active and running at 50% (checked in outputs and physically)

kizniche, 2024-03-14:
Please elaborate on the two times you mentioned a graph. The PID widget has never had a graphing element. I'll look into the value display.

drgrumpy, 2024-03-15:
Sorry I'll need to check on what I mean by graph on PID widget, now I'm confused, sorry.  For main graph I mean: Graph (Synchronous) [Highstock] Widget Configuration - I'll try to get a screenshot

drgrumpy, 2024-03-20:
See screenshot:

The status box shows that the colling fan pwm output is 43.75, but there is nothing showing on the graph for Cooling PWM, I expect to see a bar (green) with value of 43% (data grouping is off) similar to the lights, also earlier values are missing. This worked fine in 8.4

You can also see that the negative values are not showing in the pid widget.

[image]

kizniche, 2024-03-20:
Your output does not show a positive value, it's negative.

drgrumpy, 2024-03-20:
Thanks. Yes the CV is negative and doesn't show in the PID widget. But the duty cycle Cooling PWM on the output is positive (the fan wouldn't be runnin otherwise), so I expect to see it displayed as a positive value. This all used to 'just work' on 8.4, I believe I have set up everything the same.

drgrumpy, 2024-03-25:
Same type of set up - negative values display in PID widget, and Cooling (Fan) PWM output displaying as expected in 8.12.9

[image]

Also apologies about the graph confusion - I had set up mini-graphs to display the PID values

kizniche, 2024-03-25:
Thanks for the documentation. I'll investigate soon.

kizniche, 2024-03-31:
I committed a fix, if you'd like to test. It appears to be working on my end.

drgrumpy, 2024-04-16:
Apologies for being slow, I made the change in widget_pid.py line 408 and restarted backend:
`   if (document.getElementById(name + '-' + widget_id)) {`
also tried (more recent):
`   if (data[name][1] != null && document.getElementById(name + '-' + widget_id)) {`

But no change - values still show as null

kizniche, 2024-04-16:
Widgets need Widget HTML to be regenerated for any changes to take effect. This can be done on the Diagnostic Configuration page.

drgrumpy, 2024-04-16:
Ah, Okay. Just done and now showing the negative values as expected. PWM still not showing in the synchronous graph, but I guess that is a completely different issue?

closed 2025-04-26 by kizniche