kizniche/Mycodo #1377: Reverse pumping using trigger function with volume not permitted

pe2269, 2024-04-18:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[Hi, 
I want to dispense a volume of solution with an Atlas scientific pump in both directions of rotation.
I using the trigger duration function with the Output: Volume to do that.
But when i want to indicate a reverse direction of rotation with a negative volume i obtain the error :
error : Volume must be a positive value. 
Whereas in the output of my pump i must indicate a negative volume to have an inverse direction of rotation.
thanks for your help  ]*

### Versions:

 - Mycodo Version: *[8.15.13]*
 - Raspberry Pi Version: *[4B+]*
 - Raspbian OS Version: *[bullseye V11]*

pe2269, 2024-09-19:
I resolved the problem with a custom Output : Volume that accept negative values

kizniche, 2024-09-20:
Sorry for the delay. This is a good fix to the issue. I'll commit the change to the Volume Output Action that will be incorporated into the next release.