kizniche/Mycodo #1378: Upgrade problem

TeaPot169, 2024-05-03:
Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 2 Model B Rev 1.1

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Feb 12 2020 12:38:08
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 279, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

kizniche, 2024-05-03:
If it can't get the version, your system does not have internet access.

kizniche, 2024-05-03:
Duplicate of #1266.

closed 2024-05-03 by kizniche

TeaPot169, 2024-05-06:
Hi Kyle

Thank you for your response on this.
The Pi does have internet access (I can browse via chromium browser).
I have also tried entering this fix:
sudo sed -i 's/v\.\*(\\d\\\.\\d\\\.\\d)/v(\\d+)\\\.(\\d+)\\\.\\d+/g'
~/Mycodo/mycodo/utils/github_release_info.py && sudo service mycodoflask
restart

But it gives an error. I will try it again this afternoon but I am not sure
if there is something else I need to add to this command to make it work?

Kind regards and thanks in advance.

Theo

TeaPot169, 2024-05-07:
I have managed to get the update running, but got an issue during the
update. That resulted in a gateway error 502.
I still do have internet access.
After this I tried reinstalling a backup.
Here is my log of what is currently going wrong.
Please let me know how this could be resolved. I have tried rebooting as
well.

***@***.***:~ $ journalctl -u mycodoflask | tail -n 50
May 07 18:32:02 raspberrypi python[4678]: [2024-05-07 18:32:02 +0100]
[4690] [INFO] Worker exiting (pid: 4690)
May 07 18:32:04 raspberrypi python[4678]: [2024-05-07 18:32:04 +0100]
[4678] [INFO] Shutting down: Master
May 07 18:32:04 raspberrypi python[4678]: [2024-05-07 18:32:04 +0100]
[4678] [INFO] Reason: Worker failed to boot.
May 07 18:32:04 raspberrypi systemd[1]: mycodoflask.service: Main process
exited, code=exited, status=3/NOTIMPLEMENTED
May 07 18:32:04 raspberrypi systemd[1]: mycodoflask.service: Failed with
result 'exit-code'.
May 07 18:32:56 raspberrypi systemd[1]: Started Mycodo web user interface.
May 07 18:32:58 raspberrypi python[4917]: [2024-05-07 18:32:58 +0100]
[4917] [INFO] Starting gunicorn 20.1.0
May 07 18:32:58 raspberrypi python[4917]: [2024-05-07 18:32:58 +0100]
[4917] [INFO] Listening at: unix:/usr/local/mycodoflask.sock (4917)
May 07 18:32:58 raspberrypi python[4917]: [2024-05-07 18:32:58 +0100]
[4917] [INFO] Using worker: gthread
May 07 18:32:58 raspberrypi python[4917]: [2024-05-07 18:32:58 +0100]
[4932] [INFO] Booting worker with pid: 4932
May 07 18:33:10 raspberrypi python[4917]: [2024-05-07 18:33:10 +0100]
[4932] [ERROR] Exception in worker process
May 07 18:33:10 raspberrypi python[4917]: Traceback (most recent call last):
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/arbiter.py",
line 589, in spawn_worker
May 07 18:33:10 raspberrypi python[4917]:     worker.init_process()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py",
line 92, in init_process
May 07 18:33:10 raspberrypi python[4917]:     super().init_process()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/workers/base.py",
line 134, in init_process
May 07 18:33:10 raspberrypi python[4917]:     self.load_wsgi()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/workers/base.py",
line 146, in load_wsgi
May 07 18:33:10 raspberrypi python[4917]:     self.wsgi = self.app.wsgi()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/app/base.py",
line 67, in wsgi
May 07 18:33:10 raspberrypi python[4917]:     self.callable = self.load()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py",
line 58, in load
May 07 18:33:10 raspberrypi python[4917]:     return self.load_wsgiapp()
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py",
line 48, in load_wsgiapp
May 07 18:33:10 raspberrypi python[4917]:     return
util.import_app(self.app_uri)
May 07 18:33:10 raspberrypi python[4917]:   File
"/var/mycodo-root/env/lib/python3.7/site-packages/gunicorn/util.py", line
359, in import_app
May 07 18:33:10 raspberrypi python[4917]:     mod =
importlib.import_module(module)
May 07 18:33:10 raspberrypi python[4917]:   File
"/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
May 07 18:33:10 raspberrypi python[4917]:     return
_bootstrap._gcd_import(name[level:], package, level)
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap>", line 1006, in _gcd_import
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap>", line 983, in _find_and_load
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap>", line 967, in _find_and_load_unlocked
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap>", line 677, in _load_unlocked
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap_external>", line 728, in exec_module
May 07 18:33:10 raspberrypi python[4917]:   File "<frozen
importlib._bootstrap>", line 219, in _call_with_frames_removed
May 07 18:33:10 raspberrypi python[4917]:   File
"/home/pi/Mycodo/mycodo/start_flask_ui.py", line 13, in <module>
May 07 18:33:10 raspberrypi python[4917]:     app = create_app()  #
required by the wsgi config and main()
May 07 18:33:10 raspberrypi python[4917]:   File
"/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 46, in create_app
May 07 18:33:10 raspberrypi python[4917]:     register_extensions(app)
May 07 18:33:10 raspberrypi python[4917]:   File
"/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 61, in
register_extensions
May 07 18:33:10 raspberrypi python[4917]:     app = extension_babel(app)  #
Language translations
May 07 18:33:10 raspberrypi python[4917]:   File
"/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 166, in extension_babel
May 07 18:33:10 raspberrypi python[4917]:     babel = Babel(app,
locale_selector=get_locale)
May 07 18:33:10 raspberrypi python[4917]: TypeError: __init__() got an
unexpected keyword argument 'locale_selector'
May 07 18:33:10 raspberrypi python[4917]: [2024-05-07 18:33:10 +0100]
[4932] [INFO] Worker exiting (pid: 4932)
May 07 18:33:12 raspberrypi python[4917]: [2024-05-07 18:33:12 +0100]
[4917] [INFO] Shutting down: Master
May 07 18:33:12 raspberrypi python[4917]: [2024-05-07 18:33:12 +0100]
[4917] [INFO] Reason: Worker failed to boot.
May 07 18:33:12 raspberrypi systemd[1]: mycodoflask.service: Main process
exited, code=exited, status=3/NOTIMPLEMENTED
May 07 18:33:12 raspberrypi systemd[1]: mycodoflask.service: Failed with
result 'exit-code'.

TeaPot169, 2024-05-07:
I think I need to restart the whole system from scratch and install it all
from the beginning.
It seems I have lost my database as well anyway...
Is there a way one would go about this?

TeaPot169, 2024-05-07:
It looks like the daemon status is: not running.
When checking the log I get this:

File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py",
line 2819, in first
    return self.limit(1)._iter().first()
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py",
line 2906, in _iter
    execution_options={"_sa_orm_load_options": self.load_options},
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/session.py",
line 1692, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py",
line 1631, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py",
line 326, in _execute_on_connection
    self, multiparams, params, execution_options
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py",
line 1508, in _execute_clauseelement
    cache_hit=cache_hit,
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py",
line 1863, in _execute_context
    e, statement, parameters, cursor, context
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py",
line 2044, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py",
line 207, in raise_
    raise exception
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py",
line 1820, in _execute_context
    cursor, statement, parameters, context
  File
"/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py",
line 732, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such table:
conditional
[SQL: SELECT conditional.id AS conditional_id, conditional.unique_id AS
conditional_unique_id, conditional.name AS conditional_name,
conditional.position_y AS conditional_position_y, conditional.is_activated
AS conditional_is_activated, conditional.log_level_debug AS
conditional_log_level_debug, conditional.conditional_statement AS
conditional_conditional_statement, conditional.conditional_import AS
conditional_conditional_import, conditional.conditional_initialize AS
conditional_conditional_initialize, conditional.conditional_status AS
conditional_conditional_status, conditional.period AS conditional_period,
conditional.start_offset AS conditional_start_offset,
conditional.pyro_timeout AS conditional_pyro_timeout,
conditional.use_pylint AS conditional_use_pylint,
conditional.message_include_code AS conditional_message_include_code,
conditional.custom_options AS conditional_custom_options
FROM conditional
WHERE conditional.unique_id = ?
 LIMIT ? OFFSET ?]
[parameters: ('e149f68c-c813-4b4e-bc4d-24ab698a3ccb', 1, 0)]
(Background on this error at: https://sqlalche.me/e/14/e3q8)
2024-05-07 18:40:21,102 - ERROR - mycodo.database - The Mycodo database is
locked. Trying to access again in 1 second..