kizniche/Mycodo #1382: Installing dependency `bluepy` results in `Could not build wheels for bluepy`

7schmiede, 2024-05-17:
### Describe the problem/bug

Adding the input for "Mijia LYWSD03MMC" requires a dependency "bluepy@1.3.0".
During installation the error occurs "ERROR: Could not build wheels for bluepy, which is required to install pyproject.toml-based projects"

### Versions:
 - Mycodo Version: *8.15.13*
 - Raspberry Pi Version: *5*
 - Raspbian OS Version: *Raspberry Pi OS (64-bit) Kernel version: 6.6*

### Reproducibility

1. Go to Setup > Input
2. Select "Xiaomi: Mijia LYWSD03MMC"
3. Click "Add" (Missing Dependency for `bluepy` is displayed)
4. Click Install

### Expected behavior
...

### Screenshots

[image]

7schmiede, 2024-05-17:
I fixed this problem by running this
```
sudo apt-get install libglib2.0
```

Maybe this sub dependency should be defined.

kizniche, 2024-05-21:
At one point in time, it did work without installing additional packages. If it now needs libglib2.0, we should add that to the module dependency list.

closed 2025-04-26 by kizniche