kizniche/Mycodo #1386: Please add MQTT Publish Action to Vapor Pressure Deficit Function, or a Function that can publish mult-values in JSON packet format.

LucidEye, 2024-07-01:
I would like to be able to publish the value from the VPD Function over MQTT.
I am working on building a remote WiFi display that would show temp, humidity, VPD, and Outputs on or off.
I can publish measurement data via MQTT directly from the Inputs settings, but there is no option to publish the measurement data from the VPD Function.
[image]

Another, better option, would be the ability to publish multiple measurements at once in a single MQTT topic, directly from the Inputs setup, or have a Function that allowed us to publish several measurements at once in an MQTT JSON packet format for exporting data to IoT devices such as remote displays or for integration with other automation systems like Home Assistant.

I understand it might seem redundant since the Inputs already have the MQTT publish Action, but those are single measurements for each topic, and many sensors take multiple different measurements at once, so it would be useful to be able to publish those multiple measurements in a single MQTT JSON format topic.
Thank you.
[image]

LucidEye, 2024-09-21:
I'd been thinking about this and realized there's also the fact that not all of the measurements one may select to be packaged for transport will be acquired at the same time.
So I'm guessing the function would need to pull the most recent set of whatever measurements are selected from the database, at a timed interval, which means that the remote receiving device may have some values that are a few seconds behind the actual most recent measurements?  That would be fine, in my case the remote device just a display to keep an eye on the "critical" measurements without having to have a computer turned on and a browser open.

kizniche, 2024-09-21:
It wouldn't be an issue to execute the Action when the measurement is acquired, like Inputs. This simplifies everything because a timestamp doesn't need to be included in the payload.