kizniche/Mycodo #1391: MAX31855 thermocouple input library errors

kyleluiten, 2024-07-25:
### Describe the problem/bug

After a fresh install of Mycodo on varying Pi models and OS versions, the MAX31855 UART input continues to produce a runtime error upon activation, which leads to the Mycodo Error 101 message. The runtime error differs between Pi models but always comes from the same file: `/var/mycodo-root/env/lib/python3.11/site-packages/Adafruit_GPIO/GPIO.py`.

On Pi 2:
```
2024-07-25 10:54:07,136 - ERROR - mycodo.inputs.max31855_84fde3e4 - Initialization errored 3 times; giving up. Maybe the following traceback can help diagnose the issue.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/abstract_base_controller.py", line 57, in try_initialize
    self.initialize()
  File "/home/mecomposite/Mycodo/mycodo/inputs/max31855.py", line 67, in initialize
    self.sensor = MAX31855.MAX31855(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_MAX31855/MAX31855.py", line 48, in __init__
    gpio = GPIO.get_platform_gpio()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/Adafruit_GPIO/GPIO.py", line 417, in get_platform_gpio
    import RPi.GPIO
ModuleNotFoundError: No module named 'RPi'
```

On Pi 3:
```
2024-07-25 11:15:21,329 - ERROR - mycodo.inputs.max31855_06d1f9f2 - Initialization errored 3 times; giving up. Maybe the following traceback can help diagnose the issue.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/abstract_base_controller.py", line 57, in try_initialize
    self.initialize()
  File "/home/mecomposite/Mycodo/mycodo/inputs/max31855.py", line 67, in initialize
    self.sensor = MAX31855.MAX31855(
                  ^^^^^^^^^^^^^^^^^^
  File "/var/mycodo-root/env/lib/python3.11/site-packages/Adafruit_MAX31855/MAX31855.py", line 48, in __init__
    gpio = GPIO.get_platform_gpio()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/mycodo-root/env/lib/python3.11/site-packages/Adafruit_GPIO/GPIO.py", line 426, in get_platform_gpio
    raise RuntimeError('Could not determine platform.')
RuntimeError: Could not determine platform.
```

### Versions:

 - Mycodo Version: 8.15.13
 - Raspberry Pi Version: 2B and 3B both tested.
 - Raspbian OS Version: OS Lite (64-bit) Bookworm used on model 3B, and both OS Lite (32-bit) Bookworm and OS (Legacy, 32-bit) Lite Bullseye used on model 2B. All are 2024-07-04 releases.

### Reproducibility

Fresh install of OS using Raspberry Pi imaging tool, followed by fresh install of Mycodo. All defaults accepted when installing. The only changes made within Mycodo were enabling the debug logging, and then adding the MAX31855 UART input, again with default values. Upon activating the input, the runtime error is logged three times as shown above, then `Error 101: Device not set up.` is subsequently logged. I tried looking for missing packages and updating accordingly, but it seems as though everything is here and up to date.

### Additional context
I found when this input was added in 2017/18 (issue [https://github.com/kizniche/Mycodo/issues/345](https://github.com/kizniche/Mycodo/issues/345)), but I couldn't find any related discussion here after 2018. In my own attempt to debug, I found the MAX31855 library being used is deprecated as of Oct 2019: [https://github.com/adafruit/Adafruit_Python_MAX31855](https://github.com/adafruit/Adafruit_Python_MAX31855), and the Adafruit_GPIO library also as of Sep 2019. Not sure if that plays a role in the errors, but I see that other Mycodo inputs use the newer Adafruit libraries, so thought I'd mention the observation in case it is pertinent.

kizniche, 2024-07-25:
Have you tried either of the other 2 Inputs? This is likely what needs to use used to get this working: https://pypi.org/project/adafruit-circuitpython-max31855/

kyleluiten, 2024-07-25:
I'm using the [Adafruit MAX31855 board](https://www.adafruit.com/product/269), so I don't think the other two inputs will work as this breakout doesn't involve an I2C interface; I tried loading the other inputs anyways to double check and don't think that's the path forward for my hardware.

I find your documentation fairly easy to follow, so decided to try modifying the existing input and use their newer library that you linked. We'll see how far I can get.

kizniche, 2024-07-25:
I would recommend referencing how I created other Adafruit CircuitPython inputs, since there are some specific things that go along with circuitpython libraries that I do in all those modules and will need to be incorporated into a new one.

kyleluiten, 2024-07-25:
Thanks for confirming my suspicions; I was finally able to get it running based on the [MAX31865 CircuitPython](https://github.com/kizniche/Mycodo/blob/master/mycodo/inputs/max31865_circuitpython.py) input and the updated Adafruit tutorials. Once I'm done testing, would you like it in a pull request as a new input?

kizniche, 2024-07-26:
Yes, please!

PR #1392 (2024-08-01): Add input for MAX31855 using CircuitPython library
Add CircuitPython library support for MAX31855 input. Resolves #1391.