kizniche/Mycodo #1393: DS18B20 Issue

mfittipaldi90, 2024-08-24:
### Describe the problem/bug

DS18B20 Temperature sensor stopped working overnight and got stuck at 0C. Config page for w1thermsensor shows two "00-00000000." device ID
Tried a config page for OW-Shell and device ID shows 2 different unique IDs

The input with ow-shell library provides erratic measurements ranging from 0C to 96C jumping all around.

I bought a new sensor (both from adafruit) and same behavior happens with brand new sensor. Sensors are installed with 4.7k pullup resistor between signal and VCC.

### Versions:

 - Mycodo Version: 8.15.13
 - Raspberry Pi Version: 4b
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Screenshots
[image]
[image]

### Additional context

Logs show:
ERROR - mycodo.inputs.ds18b20_076d8c59 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.

kizniche, 2024-08-24:
Please describe the details of your wiring.

mfittipaldi90, 2024-08-24:
Sorry for the bad picture and diagram. This setup was working very well for the past weeks and just stopped working overnight. 
[image]

kizniche, 2024-08-24:
What is the length of wire? Have you tried 5 VDC?

kizniche, 2024-08-24:
Have you verified the resistance of the resistor?

mfittipaldi90, 2024-08-24:
Yes, tried the 5 VDC and still same behavior
Length of the wire is the length that came with the sensor itself plus the jumper wire that was soldered in.
Tried two different resistors that are 4.7k ea (both verified with multimeter).
Just tried reflashing my rpi and installing a fresh new mycodo and still not working... now I have to reconfigure it all again :)

kizniche, 2024-08-24:
Have you tried one sensor at a time?

mfittipaldi90, 2024-08-24:
Yes, right now I have only one DS18B20 connected and still same behavior. Even swapping DS18B20 and behavior is the exact same. Keep getting phantom IDs and no data reads

kizniche, 2024-08-24:
You've enabled 1wire after reinstalling?

mfittipaldi90, 2024-08-24:
yes, it is enabled

kizniche, 2024-08-24:
Sounds like damaged hardware. Have you tried other software and hardware to communicate with the sensors?

mfittipaldi90, 2024-08-24:
Ouch, guess I will have to find a new RPi. I will try that. Thank you so much for the quick responses! Sent from my iPhone

kizniche, 2024-08-26:
I would try assigning 1wire to another pin besides 4 and testing. Do you have an arduino-compatible microcontroller?

kizniche, 2024-08-26:
And by try other software, there are many available to communicate with 1wire sensors, the ds18b20 in particular, since it's probably the most populate 1wire sensor in existence.

kizniche, 2024-08-26:
There's also the possibility you have counterfeit sensors (very common). Did you buy from a reputable vendor?

mfittipaldi90, 2024-08-26:
I bought from Adafruit… pretty reputable. I purchased a new rpi 4 to test
it out as I do not have an arduino or other microcontroller to test. I have
use for the other rpi so it will come in handy.

I tried a different gpio pin but got the same result. Digging through rpi
forums i found a similar issie that was the rpi. I’ll test the new rpi
that’s coming in soon and report back for completion so we can close this
for future reference.

Mauro Fittipaldi, PhD
786-339-2074

wbard93, 2024-08-27:
I've had various problems with the DS18B20 stopping to work on the RPi. Usually it works for a while and the just stops. At least once it started working again when I disabled the input in Mycodo and re-enabled it but usually it starts working when I reboot the RPi. I remember one instance where using RPi routine independent of Mycodo I was able to read the temp and recovered it by reimaging the SD and reloading Mycodo.

I think one time that it stopped working, the RPi was bad. I've also had broken connections that have caused it to stop working.

I suspect at least part of the problem is a bug in the RPI DS18B20 library I'm using. In my view, it's not very reliable and is likely a result of the multi tasking RPi trying to talk using a bit-bang serial interface. My thought for the future is to not use the RPi to read it and use an Arduino connected to the RPi through some interface.

kizniche, 2024-08-27:
I have several Pis reading ds18b20 in production systems that have run without issue for years. There's likely something about the setup of your systems that cause them to fail.

mfittipaldi90, 2024-08-27:
I received the new Raspberry Pi and the sensors worked right away. It was my raspberry pi all along. Not sure it can be fixed…

We can close this ticket as resolved

mfittipaldi90, 2024-08-31:
After a few days of without any issues, I had this behavior overnight. For 3 hours the temp sensor was gone and the EC/PH measurements were erratic/noisy. Even after the temperature sensor coming back online by itself, the measurements were fairly jumpy. At around 8:40 AM I performed a reboot of the raspberry pi and the measurements stabilized again. Not sure what could be causing this issue? Maybe a surge or drop in the wall outlet? I am running the 15 W Raspberry Pi power supply, and measure ~7 W usage with a power meter.

[image]

mfittipaldi90, 2024-09-20:
The power to the sensor is coming from the 3.3v or 5v pin of the raspberry pi. Do you recommend using a separate power source for it?

mfittipaldi90, 2024-09-20:
For the pH and EC sensors, I am using an analog signal isolator in between them and the raspberry pi. Not sure if there is anything like that for the 1-wire. I could try to get an isolated external power supply for them

kizniche, 2024-09-20:
Your DS sensor is submerged?

mfittipaldi90, 2024-09-20:
Yes it is. Same reservoir as the pH and EC

kizniche, 2024-09-20:
Do any erratic measurements cease when you remove any sensor or combination of sensors from the fluid?

mfittipaldi90, 2024-09-20:
I saw that when the pH and EC sensors were together in the same res without the galvanic isolator circuits. Once I added those modules in, it was smooth. The overall behavior is stable and smooth overall. Not sure if it was a surge or spike in the outlet that friend the 1-wire on the RPI and that temporary loss on the second. I did add a surge protector to the outlet just in case...
[image]