kizniche/Mycodo #1394: Mycodo is not accessible and Raspberry Pi3 B+ after lkess than 12 hours

Pierrevictor67, 2024-10-03:
### Versions:

 - Mycodo Version: Mycodo v8.16.0 
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version:  Debian GNU/Linux 12 Release:        12 last one of this moring with Pi imager

Hello,
since a week I have installed Mycodo on a Pi3 B+ with a hat UPS and a Hat relay Pi 6 from Sbcomponnets.
It has a usb SSD disk connected (ORICO SSD M.2 NVME 940 256Go USB 3.1 Gen2)

A SHT31 temperature sensor and a LCD 4x20  both on I2C with a logic converter to use 3V and 5V together.
The make measures and activate the relay connected to a fridge with minimum functionning time of 120 seconds.
Temperature target of the PID is 15°C

I reinstall all, Rpi OS this morning and Mycodo with InfluxDb1 instead of InfluxDB2.

But I still have the same error.
In DEBUG logs I can see that at an undetermined time the daemon restart (12:53 am)
After that the PID no more functions correctly, and the regulation is stopped.

I observe that after this event, I cannot connect to Rpi with SSH because Ip address is no more answering to pings.

I think that with your experience you can understand what I'm doing baddly.

But I cannot find any rational explanantion by myself.

I join all the lines of the Mycodo logs frome this morning 9:30 AM to 12:53am when the error came back.

Thank's for your help.

With best regards.
Pierre
[errors Mycodo Influxdbx1.txt](https://github.com/user-attachments/files/17249290/errors.Mycodo.Influxdbx1.txt)

jimbo762, 2024-10-03:
My first guess is power.  What's the wattage on your power supply/supplies?

Pierrevictor67, 2024-10-03:
Thank you very much for your fast reply.
The UPS hat from SB components should send 2A or 2.5 A
Pi interface through screen and Keyboard told me an alert about this. But with informations that I found it should be sufficient.
Saturday I will receive an official power supply for Rpi 3 and connect it instead of the Hat UPS 
https://learn.sb-components.co.uk/UPS-Hat-for-Raspberry-Pi

With best regards.
Pierre

Pierrevictor67, 2024-11-13:
I add a comment to document the solving of my problem.
The ORICO SSD M.2 NVME 940 256Go USB 3.1 Gen2 connected trough the usb port was unstable. First I thought because of insufficient power of the Hat UPS, but replacing it with an official power supply for RPI 3B+ I still had writing errors on the ORICO SSD M.2 NVME 940 256Go USB 3.1 Gen2

I replaced it with a Samsung SD card Pro Endurance, and now my RPI is since three weeks controlling a meat dryer with Mycodo.
All is stable and functionning well.

closed 2025-01-08 by kizniche