kizniche/Mycodo #1398: OpenCV Camera not listed

Night-Lee-D, 2024-10-31:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

### Versions:

 - Mycodo Version: 8.16.0
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Bookworm

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Plug in Generic USB Camera
1. Go to Camera Tab
2. Add OpenCv Camera
3. Select Settings on OpenCV Tab
4. Go to OpenCV Device.
5. No Device Detected.

### Expected behavior

I hoping to use the camera on /dev/video0 with the opencv cam,

kizniche, 2024-10-31:
If no devices are listed, then opencv isn't detecting your camera.

Night-Lee-D, 2024-10-31:
it seems as though it was an issue with numpy 2.1.2 and opencv

kizniche, 2025-01-08:
I'll see about updating the library versions in the next release.

closed 2025-01-08 by kizniche