kizniche/Mycodo #1399: Installing RPi Power Monitor breaks Mycodo

zannis, 2024-11-01:
### Describe the problem/bug

Installing RPi Power Monitor breaks mycodo cause theres dependency conflicts between the library and mycodo, when mycodo installs these deps for influxdb and urllib to satisfy rpi-power-monitor, all calls to get measurements from influxdb fail on urllib with :
`unexpected keyword argument 'key_key_password'`

In my case what fixed it was deleting rpi power monitor input and reinstalling mycodo so that the correct dependencies were reinstalled. During installating, the dependency conflict on influxdbclient and urllib between mycodo and rpi monitor was pointed out by the installer.

### Versions:

 - Mycodo Version: Latest
 - Raspberry Pi Version:  3B
 - Raspbian OS Version: Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### Reproducibility

1. Set up an input with any kind of measurement (in my case SCD40 sensor)
2. Check fetching measurements works
3. Install RPi power monitor
4. Restart Mycodo
5. Measurements break

### Expected behavior

Nothing breaks

### Error log
```
 File "/opt/Mycodo/env/lib/python3.11/site-packages/influxdb_client/_sync/rest.py", line 311, in POST
    return self.request("POST", url,
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/influxdb_client/_sync/rest.py", line 186, in request
    r = self.pool_manager.request(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/request.py", line 70, in request
    return self.request_encode_body(method, url, fields=fields,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/request.py", line 150, in request_encode_body
    return self.urlopen(method, url, **extra_kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/poolmanager.py", line 313, in urlopen
    conn = self.connection_from_host(u.host, port=u.port, scheme=u.scheme)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/poolmanager.py", line 229, in connection_from_host
    return self.connection_from_context(request_context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/poolmanager.py", line 240, in connection_from_context
    pool_key = pool_key_constructor(request_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/urllib3/poolmanager.py", line 105, in _default_key_normalizer
    return key_class(**context)
           ^^^^^^^^^^^^^^^^^^^^
TypeError: PoolKey.__new__() got an unexpected keyword argument 'key_key_password'
```

kizniche, 2025-01-08:
How did you install it? Via the Input module within Mycodo?

zannis, 2025-01-09:
correct