kizniche/Mycodo #1403: KP303 not powering on using a Mycodo function

Chomski99, 2024-11-22:
Hi,

I'm using a Kasa KP303 which has a fixed IP address. When using Mycodo I can power each sockets on and off no problem. But when I use a function to to power one one of the sockets on after start-up it doesn't work but if I execute manually it does work.

[image]

Picture attached. Am I missing something?

Chris

Mycodo version 8.16
Rasberry Pi 4
Raspbian Bullseye

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *[e.g. 3B+]*
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to...]*
2. *[Click on...]*
3. *[Scroll down to...]*
4. *[See error...]*
5. *[replace this list with your own text]*

### Expected behavior

*[A clear and concise description of what you expected to happen - replace this with your own text]*

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

kizniche, 2024-11-22:
Please fill out the bug report entirely, because I cannot understand how to reproduce the issue or what you believe to be the expected behavior. Edit your post and change the areas that have "replace this with your own text".

kizniche, 2024-11-24:
So your goal is simply to have it turn on the output when Mycodo starts? This can be done by setting Startup to On in the Output config.