kizniche/Mycodo #1412: Install issue, latest version, ModuleNotFoundError: No module named 'flask_babel'

gbartlett82, 2025-03-10:
### Describe the problem/bug

On 10 March 2025, ran the following: 
`curl -L https://kizniche.github.io/Mycodo/install | bash`

The following error occurs:
```
#### Generating widget HTML files
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/utils/widget_generate_html.py", line 9, in <module>
    from mycodo.config import PATH_TEMPLATE_USER
  File "/opt/Mycodo/mycodo/config.py", line 10, in <module>
    from flask_babel import lazy_gettext as lg
ModuleNotFoundError: No module named 'flask_babel'
```

#### Running initialization

I am on a raspberry pi 4b (32-bit) fully up-to-date.

### Versions:

 - Mycodo Version: latest as of 10 March 2025
 - Raspberry Pi Version: 4b
 - Raspbian OS Version:
   - Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
  
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. run install command noted above.
2. error (noted above) occurs.

### Expected behavior

Error doesn't occur and webserver comes up.

### Screenshots

### Additional context

kizniche, 2025-03-10:
If flask_babel wasn't installed, you likely had an issue downloading it during the install due to a network issue. The setup log will show this. Reattempt the install.

closed 2025-03-19 by kizniche

gbartlett82, 2025-03-26:
Following up.  I was running an older version of raspbian.  Everything loaded great once I updated.