kizniche/Mycodo #1418: 403 Forbbiden access to the login page

salex996, 2025-04-17:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

I have installed Mycodo several times, several versions but always I receive the same message when I try to log in for the first time: 403 Forbidden

[image]

### Versions:

 - Mycodo Version: 8.16.0
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Raspberry pi OS 12 64-bit

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. I install the raspberry pi os 12 64-bit
2. I run this command curl -L https://kizniche.github.io/Mycodo/install | bash
3. The system looks like the installation has been done properly but it tries to connect to localhost the 10 times
4. I try to put the IP of the raspberry in google chrome 
5. I need to click on advance options because google chrome wanrs that it is a non-safe page
6. The next page is 403 forbbiden

### Expected behavior

It should appear the page where you can set the credentials to enter in mycodo

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

I was investigating about SSL certificates, package incompatibilities and other similar bugs, but I can't find anything that can help me, can anyone please assist me a little bit?

salex996, 2025-04-17:
I tried as well with previous versions of raspberry pi, I tried with 32 bits and nothing...

pulgamecanica, 2025-04-17:
Would it be possible to have a look at your nginx log?

Usually nginx logs are founded at: `/var/log/nginx/`

For example:
```
var/log/nginx/access.log
var/log/nginx/error.log
```

salex996, 2025-04-17:
Yes for sure, 

if I run cat /var/log/nginx/access.log

[image]

salex996, 2025-04-17:
additionally in the error.log

[image]

[image]

kizniche, 2025-04-18:
Were there any errors during the install? setup.log is located in the /install directory.

salex996, 2025-04-18:
[setup.log](https://github.com/user-attachments/files/19809220/setup.log)

Please find attached the logs, it can be seen that there are several messages that may be related with the error:

[image]

Error importing pprint from marshmellow 

[image]

[image]

[image]

kizniche, 2025-04-18:
You should look for errors in the setup associated with downloading/installing dependencies. This includes apt and pip.

salex996, 2025-04-18:
even I have tried to install it in docker but the same problem appeared and actually the container mycodo_flask is not running properly

kizniche, 2025-04-18:
Try these commands:

```
sudo /opt/Mycodo/env/bin/pip install flask-marshmallow==1.3.0 && sudo /opt/Mycodo/mycodo/scripts/upgrade_post.sh
```

salex996, 2025-04-19:
I tried but I got the same result

kizniche, 2025-04-19:
What OS version are you using? You didn't specify light or desktop. Light is recommended.

salex996, 2025-04-19:
it is lite but I also tried with desktop, 32 and 64, and I tried with the previous version of raspberry pi, 32 and 64 lite and desktop...

DamnTrak, 2025-04-19:
Hello, I have exactly the same problem in the same conditions and with the same symptoms as salex996 with a rasberry pi 5 and a rasberry pi 3b, fresh install.

kizniche, 2025-04-19:
This should fix the issue:

```
sudo /opt/Mycodo/env/bin/pip install flask-marshmallow==1.3.0 marshmallow-sqlalchemy==1.4.2 && sudo /opt/Mycodo/mycodo/scripts/upgrade_post.sh
```

AoT-inc, 2025-04-20:
When installing Mycodo, a compatibility issue may occur because version 4.0 of marshmallow is installed.
You can resolve this issue by explicitly specifying the marshmallow version in requirements.txt and reinitializing the virtual environment.

Add the following line to your requirements.txt:
marshmallow==3.19.0

1. Remove the existing virtual environment: 
 rm -rf /opt/AoT/env

2. Create and activate a new virtual environment
python3 -m venv /opt/Mycodo/env
source /opt/Mycodo/env/bin/activate

3.Upgrade pip and install dependencies
cd /opt/Mycodo
pip install --upgrade pip
pip install --break-system-packages -r install/requirements.txt

gwansukpark, 2025-04-20:
When installing Mycodo, a compatibility issue may occur because version 4.0 of marshmallow is installed.
You can resolve this issue by explicitly specifying the marshmallow version in requirements.txt and reinitializing the virtual environment.

Add the following line to your requirements.txt:
marshmallow==3.19.0

1. Remove the existing virtual environment:
rm -rf /opt/AoT/env

2. Create and activate a new virtual environment
python3 -m venv /opt/Mycodo/env
source /opt/Mycodo/env/bin/activate

3. Upgrade pip and install dependencies
cd /opt/Mycodo
pip install --upgrade pip
pip install --break-system-packages -r install/requirements.txt

kizniche, 2025-04-20:
Excuse my slight detour from the actual issue at hand, but I noticed a license violation just now and wanted to address. The last commenter (@gwansukpark) appears to have unintentionally posted with an account that has copied and improperly attributed my software. Perhaps they don't realize that an email is sent when a comment is made:

[image]

This initial comment was swiftly deleted. I can only presume because the account it was posted from was @AoT-Inc and the poster forgot to change one of the three instances of the install directory from "/opt/AoT". This user has a repo at https://github.com/AoT-inc/AoT that is a copy of Mycodo. This isn't an issue by itself, as the license I released Mycodo under is GPL3, however, this license requires the original attributes to remain intact, meaning my name and contact information. Taking a quick look at one of the controller classes, we can see there was obviously a find-replace of "Mycodo" with "AoT" as well as my email address with their own email address:

https://github.com/AoT-inc/AoT/blob/cb5ab87c67ae91996120eb73b641ec92d830f78d/aot/controllers/controller_pid.py#L3-L23

```pytthon
# controller_pid.py - PID controller that manages discrete control of a
#                     regulation system of inputs, outputs, and devices
#
#  Copyright (C) 2015-2020 Kyle T. Gabriel <aot@aot-inc.com>
#
#  This file is part of AoT
#
#  AoT is free software: you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation, either version 3 of the License, or
#  (at your option) any later version.
#
#  AoT is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
#  GNU General Public License for more details.
#
#  You should have received a copy of the GNU General Public License
#  along with AoT. If not, see <http://www.gnu.org/licenses/>.
#
#  Contact at aot-inc.com
```

Preserving the current comment for posterity:

[image]

@gwansukpark and @AoT-inc, what do you have to say about this?

kizniche, 2025-04-20:
Also, anyone reading https://github.com/kizniche/Mycodo/issues/1418#issuecomment-2817186307, do not follow those instructions. It is clear the poster does not understand the system. Again, do not delete your virtual environment. Follow my instructions instead (https://github.com/kizniche/Mycodo/issues/1418#issuecomment-2816804048) and report back whether the fix was successful, so it can be included in the next release.

gwansukpark, 2025-04-20:
I’m very sorry.

While creating the project with the help of AI for testing purposes, everything ended up being changed automatically.
I will carefully review and reflect on the points you mentioned, and make sure the original authorship is properly credited and that there are no copyright issues.

In hindsight, I should have waited until it was more complete before making it public.
Since it was just a test upload today, I wasn’t able to check everything thoroughly.

For now, I will take down the GitHub repository and re-upload it after making all the necessary corrections you pointed out.

salex996, 2025-04-21:
Hi, 

Just come here to confirm that the solution provided worked

Running sudo /opt/Mycodo/env/bin/pip install flask-marshmallow==1.3.0 marshmallow-sqlalchemy==1.4.2 && sudo service mycodoflask restart the problem dissapeared at least for me, 

Thank you very much!!!

flo269, 2025-04-25:
> Hi,
> 
> Just come here to confirm that the solution provided worked
> 
> Running sudo /opt/Mycodo/env/bin/pip install flask-marshmallow==1.3.0 marshmallow-sqlalchemy==1.4.2 && sudo service mycodoflask restart the problem dissapeared at least for me,
> 
> Thank you very much!!!

The same worked for me here. Thanks!!

closed 2025-04-26 by kizniche