kizniche/Mycodo #1421: Mycodo Install fails on Installing and configuring nginx web server

spilsbury4, 2025-04-27:
### Describe the problem/bug

Mycodo v8.16.1 install fails at Installing and configuring nginx web server. 

### Versions:

 - Mycodo Version: v8.16.1
 - Raspberry Pi Version: Compute Module 4 Rev 5 CM4104008 (SC0672); RASPBERRY PI CM4IO BOARD (SC0326)
 - Raspbian OS Version: Raspberry Pi OS Lite (64-bit) (Released 2024-11-19)

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Fresh flash (to eMMC) of Raspberry Pi OS Lite (64-bit) with SSH enabled
2. SSH into Pi
3. Quick Install Mycodo (curl -L https://kizniche.github.io/Mycodo/install | bash)
4. See error... full log file attached.

[setup.log](https://github.com/user-attachments/files/19932326/setup.log)

### Expected behavior

Expect successful installation of Mycodo v8.16.1

### setup.log

#### Installing and configuring nginx web server

#### Disabling service for nginx web server
Failed to disable unit: Unit file mycodoflask.service does not exist.

#### Enabling service for nginx web server
Synchronizing state of nginx.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable nginx
Created symlink /etc/systemd/system/mycodoflask.service → /opt/Mycodo/install/mycodoflask.service.
Created symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service → /opt/Mycodo/install/mycodoflask.service.

#### Restarting nginx
#### Reloading mycodoflask
mycodoflask.service is not active, cannot reload.

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...

kizniche, 2025-04-27:
I don't see any errors. What is the issue?

spilsbury4, 2025-04-28:
I was having an issue loading the web server and was thinking the install was stalling and that there was an error related to reloading the mycodoflask.service, but I just checked again and now it is working... sorry for the issue opened I will close!!

And thanks for confirming that there is no error in the log - I appreciate all the amazing work you have done with this!!

closed 2025-04-28 by spilsbury4

kizniche, 2025-04-28:
No problem. There just wasn't a clear explanation of the issue, so just wondering what exactly you were experiencing that wasn't working. I have it attempt to connect up to 10 times (see "up to 10 times" in the last line) because slower computers can take a while for the web server to start. 10 is overkill (10 attempts over 10 minutes), but it works.